Azure/iot-edge-opc-publisher-v1 #124: Error when trying to make it work as an IoT Edge module on a Raspberry PI3

ArnMl, 2018-07-04:
Hello,

As the title says, I'm trying to use this module with IoT Edge on a Raspberry Pi3.

I couldn't use directly the public repository because of the "exec format error" as explained in the issue #112.
I built my own image of the code with the Dockerfile.arm32v7 and uploaded it to an Azure docker repository, then set it as a module for my Iot Edge instance.

First, the logs showed that it couldn't find the connection string, so I passed it as an environment variable.
After that, I got the message `ErrorCode:IotHubUnauthorizedAccess;Unauthorized` and I can't find the problem.
The connection string is correct, the clock of the raspberry pi is correctly set and synchronised.

Full error message : 
``07/02/2018 14:23:05:{"Message":"ErrorCode:IotHubUnauthorizedAccess;Unauthorized","ExceptionMessage":"Tracking ID:edf5471e3e694686a8d8bab84169bb4c-G:16-TimeStamp:07/02/2018 14:23:05"}
07/02/2018 14:23:05:    at Microsoft.Azure.Devices.HttpClientHelper.ExecuteAsync(HttpClient httpClient, HttpMethod httpMethod, Uri requestUri, Func`3 modifyRequestMessageAsync, Func`2 isMappedToException, Func`3 processResponseMessageAsync, IDictionary`2 errorMappingOverrides, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.HttpClientHelper.GetAsync[T](Uri requestUri, TimeSpan operationTimeout, IDictionary`2 errorMappingOverrides, IDictionary`2 customHeaders, Boolean throwIfNotFound, CancellationToken cancellationToken)
   at OpcPublisher.IotHubMessaging.InitAsync() in /build/src/IotHubMessaging.cs:line 183``

Thanks.

hansgschossmann, 2018-07-07:
the changes for IoTEdge support will come in the next few days to latest. it would be great if you could validate them on Pi3 after they are in.

hansgschossmann, 2018-07-13:
Can you give it a try after this PR https://github.com/Azure/iot-edge-opc-publisher/pull/131 has been merged?

hansgschossmann, 2018-11-06:
a container for ARM32 is now available via Microsoft Container Registry. find the tags available on https://hub.docker.com/r/microsoft/iot-edge-opc-publisher/

closed 2018-11-06 by hansgschossmann