Azure/iot-edge-opc-publisher-v1 #128: generate docker image on raspberryPI 3

eoursel, 2018-07-11:
Hi,

I tried to generate the docker image on RaspberryPI with your last dockerfile (yesterday). 

In the building process the command adduser fails. 

`# Add an unprivileged user account for running the module
RUN adduser -Ds /bin/sh moduleuser`

Looks like the -Ds option is not recognized. I removed the option -Ds but when iotedge starts there is an error to create the certificate. I used a private Container Registry in Azure to host the container. The installation process with Hub Iot Edge is ok. 

`[20:50:51 INF] Current directory is: /app
[20:50:52 INF] Log file is: publisher-publisher.log
[20:50:52 INF] Log level is: info
[20:50:52 INF] Publisher is starting up...
[20:50:52 INF] Application Certificate store type is: X509Store
[20:50:52 INF] Application Certificate store path is: CurrentUser\UA_MachineDefault
[20:50:52 INF] Application Certificate subject name is: publisher
[20:50:52 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[20:50:52 INF] with a 2048 bit key and 256 bit hash.
[20:51:03 INF] Application certificate is for Application URI 'urn:publisher:publisher:microsoft:', Application 'publisher and has Subject 'publisher'
[20:51:03 INF] Trusted Issuer store type is: Directory
[20:51:03 INF] Trusted Issuer Certificate store path is: CertificateStores/issuers
[20:51:03 INF] Trusted Peer Certificate store type is: Directory
[20:51:03 INF] Trusted Peer Certificate store path is: CertificateStores/trusted
[20:51:03 INF] Rejected certificate store type is: Directory
[20:51:03 INF] Rejected Certificate store path is: CertificateStores/rejected
[20:51:03 INF] Rejection of SHA1 signed certificates is disabled
[20:51:03 INF] Minimum certificate key size set to 1024
[20:51:03 INF] Adding publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
[20:51:03 ERR] Can not add publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
System.UnauthorizedAccessException: Access to the path '/app/CertificateStores/trusted' is denied. ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at System.IO.FileSystem.CreateDirectory(String fullPath)
   at Opc.Ua.DirectoryCertificateStore.WriteFile(Byte[] data, String fileName, Boolean includePrivateKey)
   at Opc.Ua.DirectoryCertificateStore.Add(X509Certificate2 certificate, String password)
   at OpcPublisher.OpcStackConfiguration.ConfigureAsync() in /app/opcpublisher/OpcStackConfiguration.cs:line 223
`
Regards

eoursel, 2018-07-12:
Same issue on amd64. The CertificateStores folder can't be created since the user moduleuser do not have the write permission to  create the folder /app/CertificateStores. 

> 10:06:13 ERR] Can not add publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
System.UnauthorizedAccessException: Access to the path '/app/CertificateStores/trusted' is denied. ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at System.IO.FileSystem.CreateDirectory(String fullPath)
   at Opc.Ua.DirectoryCertificateStore.WriteFile(Byte[] data, String fileName, Boolean includePrivateKey)
   at Opc.Ua.DirectoryCertificateStore.Add(X509Certificate2 certificate, String password)
   at OpcPublisher.OpcStackConfiguration.ConfigureAsync() in /app/opcpublisher/OpcStackConfiguration.cs:line 223

>

hansgschossmann, 2018-07-13:
i am on this one right now and will update you later today.

hansgschossmann, 2018-07-13:
PR is on the way. Pls verify when it is merged.

eoursel, 2018-07-13:
Hi @hansgschossmann 

I am testing your PR on the RPI3. Thank you very much for providing this patch. 

When I start the container with
`docker run -it -v /iotedge/certs:/home/moduleuser/.dotnet/corefx/cryptography/x509stores opcpublisher.arm32:latest opcpublishertest $iothub`

The container starts (good news ...)

> [20:17:41 INF] Current directory is: /appdata
[20:17:42 INF] Log file is: eb66f17b42e5-publisher.log
[20:17:42 INF] Log level is: info
[20:17:42 INF] Publisher is starting up...
[20:17:42 INF] Application Certificate store type is: X509Store
[20:17:42 INF] Application Certificate store path is: CurrentUser\UA_MachineDefault
[20:17:42 INF] Application Certificate subject name is: opcpublishertest
[20:17:42 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[20:17:42 INF] with a 2048 bit key and 256 bit hash.
[20:17:49 INF] Application certificate is for Application URI 'urn:eb66f17b42e5:opcpublishertest:microsoft:', Application 'opcpublishertest and has Subject 'opcpublishertest'
[20:17:49 INF] Trusted Issuer store type is: Directory
[20:17:49 INF] Trusted Issuer Certificate store path is: CertificateStores/issuers
[20:17:49 INF] Trusted Peer Certificate store type is: Directory
[20:17:49 INF] Trusted Peer Certificate store path is: CertificateStores/trusted
[20:17:49 INF] Rejected certificate store type is: Directory
[20:17:49 INF] Rejected Certificate store path is: CertificateStores/rejected
[20:17:49 INF] Rejection of SHA1 signed certificates is disabled
[20:17:49 INF] Minimum certificate key size set to 1024
[20:17:49 INF] Adding publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
[20:17:49 INF] OperationTimeout set to 120000
[20:17:49 INF] Publisher server base address: opc.tcp://eb66f17b42e5:62222/UA/Publisher
[20:17:49 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[20:17:49 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
[20:17:49 INF] opcstacktracemask set to: 0x0
[20:17:50 INF] There is no site configured.
[20:17:50 INF] Publisher configured to not auto trust server certificates. When connecting to servers, you need to manually copy the rejected server certs to the trusted store to trust them.
[20:17:50 INF] Starting server on endpoint opc.tcp://eb66f17b42e5:62222/UA/Publisher ...
[20:17:55 INF] Server started.
[20:17:55 INF] Using default telemetry configuration.
[20:17:55 INF] The name of the configuration file for published nodes is: /appdata/publishednodes.json
[20:17:55 INF] The node configuration file '/appdata/publishednodes.json' does not exist. Continue and wait for remote configuration requests.
[20:17:55 INF] There are 0 nodes to publish.
[20:17:55 INF] IoTHub device cert store type is: X509Store
[20:17:55 INF] IoTHub device cert path is: My
[20:17:55 INF] Attempting to register ourselves with IoT Hub using owner connection string.
[20:17:59 INF] Device 'opcpublishertest' found in IoTHub registry. Remove it.
[20:17:59 INF] Adding device 'opcpublishertest' to IoTHub registry.
[20:17:59 INF] Generate device connection string.
[20:17:59 INF] Adding device connectionstring to secure store.
[20:18:11 INF] Attempting to read device connection string from cert store using subject name: opcpublishertest
[20:18:11 INF] IoTCentral mode: False
[20:18:13 INF] Connection status changed to 'Connected', reason 'Connection_Ok'
[20:18:14 INF] Message processing and hub communication configured with a send interval of 10 sec and a message buffer size of 262144 bytes.
[20:18:14 INF] Creating task process and batch monitored item data updates...
[20:18:14 INF]
[20:18:14 INF]
[20:18:14 INF] Publisher is running. Press CTRL-C to quit.

If I start the container with a named docker volume, the container does not start and I don't understand why we have an issue with that. 

` docker run -it -v iotedge_certstore:/home/moduleuser/.dotnet/corefx/cryptography/x509stores opcpublisher.arm32:latest opcpublishertest $iothub
`

> [20:22:34 INF] Current directory is: /appdata
[20:22:34 INF] Log file is: 55fb6f99d0d3-publisher.log
[20:22:34 INF] Log level is: info
[20:22:34 INF] Publisher is starting up...
[20:22:35 INF] Application Certificate store type is: X509Store
[20:22:35 INF] Application Certificate store path is: CurrentUser\UA_MachineDefault
[20:22:35 INF] Application Certificate subject name is: opcpublishertest
[20:22:35 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[... 52 lines cut ...]
   at System.IO.FileSystem.CreateDirectory(String fullPath)
   at System.IO.Directory.CreateDirectory(String path)
   at Internal.Cryptography.Pal.DirectoryBasedStoreProvider.AddCertToStore(ICertificatePal certPal)
   at Internal.Cryptography.Pal.DirectoryBasedStoreProvider.Add(ICertificatePal certPal)
   --- End of inner exception stack trace ---
   at Internal.Cryptography.Pal.DirectoryBasedStoreProvider.Add(ICertificatePal certPal)
   at System.Security.Cryptography.X509Certificates.X509Store.Add(X509Certificate2 certificate)
   at Opc.Ua.X509CertificateStore.Add(X509Certificate2 certificate, String password)
   at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, [... cut ...]
System.UnauthorizedAccessException: Access to the path '/home/moduleuser/.dotnet/corefx/cryptography/x509stores/ua_machinedefault' is denied. ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
   at System.IO.FileSystem.CreateDirectory(String fullPath)
   at System.IO.Directory.CreateDirectory(String path)
   at Internal.Cryptography.Pal.DirectoryBasedStoreProvider.AddCertToStore(ICertificatePal certPal)
   at Internal.Cryptography.Pal.DirectoryBasedStoreProvider.Add(ICertificatePal certPal)
[20:26:31 FTL] Publisher exiting...

Have a nice weekend. 
Eric

hansgschossmann, 2018-07-22:
I have meanwhile remove the moduleuser alltogether and run the module as root. Can you give it a try again?

eoursel, 2018-07-22:
Hi @hansgschossmann 

Yes, I made some tests  but unfortunately, my Azure subscription was closed (because of a contract issue with Microsoft) before finishing the  tests on AMD64 and ARM32.  I have to solve this administrative issue to continue testing.  

I understand what you have done, Nevertheless, just a question. Running the server as a non privileged account was perhaps a good idea from a cyber security point of view. 

Thanks again. 

Eric

hansgschossmann, 2018-07-22:
I decided to remove the non priviledged account for now for the following reasons:

  *   The only way to get elevation of privileges is through calling OPC methods or IoTHub methods, which both are only allowed after authentication.
  *   There is no check by IoTEdge if the module runs non-priviledged and it is up to the module vendor to decide how the module runs.
  *   When module integrity checking is in place and enabled, then only elevation of privileges attack are the attack surface, which are protected as above.

Still it would be nice to run as non-privileged. If I have time this is something to enable and chmod all the different paths OPC Publisher requires access to.

Thx
Hans

hansgschossmann, 2018-11-06:
An ARM image is not available via Microsoft Container Registry. Pls find the tags at https://hub.docker.com/r/microsoft/iot-edge-opc-publisher/

closed 2018-11-06 by hansgschossmann