Azure/iot-edge-opc-publisher-v1 #130: Docker build error in windows

Norman03, 2018-07-12:
Can anyone say why the highlighted(yellow) warnings displayed while building the docker?

[image]

I have created a windows docker and created a linux container inside windows docker. 

command for docker build: (cmd)
docker build –f Dockerfile -t pensive_jones . 

**command for docker run: (cmd)**
docker run pensive_jones testing HostName=xxx.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=DY898xsafsfWxqUT6ytgRSAFSDGSDF/tUZU=

hansgschossmann, 2018-07-13:
the yellow marked lines are not from the docker build, but from the docker run command.

publisher does use a configuration file to load the nodes to publish (this is one option). this file is not there, so you either need to provide it or use iothub direct method calls to configure publisher. pls check out the readme of the repo and let me know if the description is not good enough.

Norman03, 2018-07-13:
Thank you! I tried by giving this option **--pf="<Absolute path of publishednodes.json >"** while using docker run command. Even then I am getting the same warning. Could you help me on the same?

hansgschossmann, 2018-07-13:
the permission denied issue you see in the log output, should go away with this PR: https://github.com/Azure/iot-edge-opc-publisher/pull/131

Norman03, 2018-07-13:
Nice to see. How about publishernodes.json missing warning? Please help me in this.

hansgschossmann, 2018-07-13:
This is actually not a warning but a information. Are you planning to use a preconfigured .json file? If yes, then most likely your configuration is not correct.

Norman03, 2018-07-13:
I went through my json file and tested by running in native windows application. The application run successfully. But when I run in docker container I use the same json file I get this error.
Can you please put the exact commands you used to run in docker container?

hansgschossmann, 2018-07-13:
I think now i understand. You need to add a -v mapping option to your docker run. With the latest it should look something like: docker run -v d:\iotedge:/appdata iot-edge-opc-publisher ....
Assuming your publishednodes.json is in d:\iotedge

Norman03, 2018-07-14:
Thank you! @hansgschossmann 
I will check the -v option and reply back in next 2 hours. Can you please suggest me few links for creating a docker linux container?

Norman03, 2018-07-15:
Great. -v works fine. Let me post the exact steps I followed to run the opcpublisher application in the docker container. I hope this helps you to update in the readme.md

hansgschossmann, 2018-11-06:
No activity for a while. Closing for now. Pls reopen if required.

closed 2018-11-06 by hansgschossmann