Azure/iot-edge-opc-publisher-v1 #138: how to prevent Publisher from adding 3 backslashes to all JSON strings

designncode7, 2018-07-24:
This is what Publisher sends to IoTHub and then to Azure Stream Analytics:

Thanks.

{"NodeId":"nsu=http://example.proof.com;i=2","ApplicationUri":"urn:freeopcua:python:server","DisplayName":"OpcuaMsg","Value":{"Value":"{ \\\"sensordata\\\": [ {\\\"msrname\\\": \\\"f4055\\\", \\\"msrvalue\\\": 150, \\\"createdon\\\": \\\"2018-07-23 17:19:45.051770\\\" },{\\\"msrname\\\": \\\"p4072\\\", \\\"msrvalue\\\": 106, \\\"createdon\\\": \\\"2018-07-23 17:19:45.051770\\\" },{\\\"msrname\\\": \\\"v4062\\\", \\\"msrvalue\\\": 177, \\\"createdon\\\": \\\"2018-07-23 17:19:45.051770\\\" },{\\\"msrname\\\": \\\"v4069\\\", \\\"msrvalue\\\": 129, \\\"createdon\\\": \\\"2018-07-23 17:19:45.051770\\\" },{\\\"msrname\\\": \\\"t4012\\\", \\\"msrvalue\\\": 166, \\\"createdon\\\": \\\"2018-07-23 17:19:45.051770\\\" },{\\\"msrname\\\": \\\"t4015\\\", \\\"msrvalue\\\": 108, \\\"createdon\\\": \\\"2018-07-23 17:19:45.052731\\\" },{\\\"msrname\\\": \\\"t4018\\\", \\\"msrvalue\\\": 114, \\\"createdon\\\": \\\"2018-07-23 17:19:45.052731\\\" },{\\\"msrname\\\": \\\"t4078\\\", \\\"msrvalue\\\": 164, \\\"createdon\\\": \\\"2018-07-23 17:19:45.052731\\\" },{\\\"msrname\\\": \\\"t8311\\\", \\\"msrvalue\\\": 118, \\\"createdon\\\": \\\"2018-07-23 17:19:45.053723\\\" },{\\\"msrname\\\": \\\"t8321\\\", \\\"msrvalue\\\": 92, \\\"createdon\\\": \\\"2018-07-23 17:19:45.053723\\\" },{\\\"msrname\\\": \\\"t8331\\\", \\\"msrvalue\\\": 117, \\\"createdon\\\": \\\"2018-07-23 17:19:45.054243\\\" },{\\\"msrname\\\": \\\"t8341\\\", \\\"msrvalue\\\": 61, \\\"createdon\\\": \\\"2018-07-23 17:19:45.054737\\\" },{\\\"msrname\\\": \\\"t8371\\\", \\\"msrvalue\\\": 86, \\\"createdon\\\": \\\"2018-07-23 17:19:45.054737\\\" },{\\\"msrname\\\": \\\"t8381\\\", \\\"msrvalue\\\": 119, \\\"createdon\\\": \\\"2018-07-23 17:19:45.055235\\\" },{\\\"msrname\\\": \\\"t8551\\\", \\\"msrvalue\\\": 181, \\\"createdon\\\": \\\"2018-07-23 17:19:45.055739\\\" },{\\\"msrname\\\": \\\"t8561\\\", \\\"msrvalue\\\": 183, \\\"createdon\\\": \\\"2018-07-23 17:19:45.056238\\\" } ]}","SourceTimestamp":"2018-07-23T15:19:45.0657350Z"},"EventProcessedUtcTime":"2018-07-23T15:22:36.1538872Z","PartitionId":3,"EventEnqueuedUtcTime":"2018-07-23T15:20:35.3670000Z","IoTHub":{"MessageId":null,"CorrelationId":null,"ConnectionDeviceId":"opcuadevice2","ConnectionDeviceGenerationId":"636673452149025346","EnqueuedTime":"2018-07-23T15:20:35.3750000Z","StreamId":null}}

hansgschossmann, 2018-07-27:
What you mean by 3 backslashes? Your data you are publishing contains double quotes, which have to be escaped.

hansgschossmann, 2018-11-06:
No response for a while. Closing for now. Pls reopen if required.

closed 2018-11-06 by hansgschossmann