Azure/iot-edge-opc-publisher-v1 #146: What about the publication of Events? (feature request)

eoursel, 2018-08-10:
The OPC UA PubSub specification supports the publication of OPC UA Events. In the Part 14 specification, there is the PublishedEventsType. 

Do you plan to support the publication of events in the OPC Publisher? 

As a matter of fact, most industrial processes generate Alarms. This feature is really important for applications which need to analyze and store alarms with Azure services. 

Best regards,

Norman03, 2018-08-10:
@eoursel  I just find this device in azure IoT catalog. They have pubsub running with open source OPC UA. Hope this helps!
(https://catalog.azureiotsolutions.com/details?title=IIoT-Starter-Kit )

eoursel, 2018-08-16:
Hi @Norman03 

Thank you for the information. Nevertheless, the support of the PubSub into open62541 seems to be just at the very beginning. The OPCUA PubSub is a very large specification with many optional facets. Nobody on the market (commercial or open source) is currently supporting the complete PubSub specification.  

My question is focused on this open source project which is very interesting to connect OPC UA servers on Azure. So i would like to understand  what is the roadmap regarding the compliance of the OPC Publisher with OPC UA PubSub specification. I know that we will not have everything in one day, but  did you define some priorities and a roadmap?  

Regards