Azure/iot-edge-opc-publisher-v1 #147: IotEdge Agent not able to deploy specified module 'Error calling Create module iot-edge-opc-publisher: Container runtime error'

manuelfuchs, 2018-08-17:
I declared the iot-edge-opc-publisher module just like declared in the tutorial, but when the agent on the edge device tries to deploy the module. the agent gets a exception.

[image]

I declared the module via the 'Set Modules' in the IoT Hub for this specific edge device and the edge device is a azure windows 10 virtual machine.

This were my container creation options ...
[image]
... first and then tried exactly yours from the instructions, neither did work.

hansgschossmann, 2018-11-06:
I have updated the documentation for use with the GA version of IoT Edge a few weeks ago. Can you please retry with those?

swapshahir, 2018-11-12:
I am facing same issue while deploying OPC publisher as module in azure IOT edge. Is there any solution for this?

hansgschossmann, 2019-01-29:
This should work without issues with the updated documentation and latest IoT Edge. There are now OPC Publisher containers available for Windows/AMD64 and Linux/ARM as well. Pls reopen if you still see an issue.

closed 2019-01-29 by hansgschossmann