Azure/iot-edge-opc-publisher-v1 #148: Error Message in Connected Factory:  Error while connecting to the OPC UA server

matteo-bon, 2018-08-25:
I am trying to connect a OPCUA server to Azure using Connected Factory, following this [guide](https://docs.microsoft.com/it-it/azure/iot-accelerators/iot-accelerators-connected-factory-gateway-deployment). 

However I'm facing the same issue of #58: everything works until section `Enable your gateway`, when I get the error `Error while connecting to the OPC UA server`:
[image]. 

As suggested in #58 I added entries to `/etc/hosts` file:
`192.168.43.76	     publisher.matt14.io  # for gateway`
`192.168.43.225    opcuasrv.maci14.io # for opcua server`

Then I start publisher and proxy with: 

- for publisher `sudo docker run --rm -it -v /SharedFolder:/docker -v x509certstores:/root/.dotnet/corefx/cryptography/x509stores --network iot_edge --name publisher -h publisher -p 62222:62222 --add-host opcuasrv.maci14.io:192.168.43.225 microsoft/iot-edge-opc-publisher:2.1.3 publisher "<iothubconnectionstring>" --lf /docker/publisher.log.txt --as true --si 1 --ms 0 --tm true --vc true --di 30
 `
- for proxy `sudo docker run -it --rm -v /SharedFolder:/mapped --network iot_edge --name proxy --add-host opcuasrv.maci14.io:192.168.43.225 microsoft/iot-edge-opc-proxy:1.0.2 -i -c "<iothubconnectionstring>" -D /mapped/cs.db` and 
- `sudo docker run -it --rm -v /SharedFolder:/mapped --network iot_edge --name proxy --add-host opcuasrv.maci14.io:192.168.43.225 microsoft/iot-edge-opc-proxy:1.0.2 -D /mapped/cs.db`

Publisher and proxy seems to work without errors, but in `/SharedFolder/CertificateStores` there is no rejected certificates to move, so It seems that the Client in the web dashboard is not able to communicate with the publisher on my PC. 

I tried also to connect with ` opc.tcp://publisher.matt14.io:62222` as suggested in the previous issue, but the error remains (same error also trying to connect to the server _opcuasrv.maci14.io_). 

Any suggestions?
Thank you!

shacotaco, 2018-10-29:
Did you find a way to solve this? 
I am having the exact same issue...

ggokka, 2018-10-30:
@shacotaco How about opc.tcp://publisher.matt14.io:62222

ggokka, 2018-10-30:
@matteo-bon Did you check whether your service(opc.tcp://publisher.matt14.io:62222) is fine in the UaEpert tool.

hansgschossmann, 2018-11-07:
A couple of things:
- Please use the latest Connected Factory. I have added a few fixes in this area in August release and August patches
- Use iot-edge-opc-proxy:1.0.4 (the documentation is updated meanwhile)
- Watch the log output of the proxy
- When you press the connect button, you should see a proxy log output which try to resolve the hostname in the URL you entered. In your case "publisher". 
- Then you should see more package exchanges and a context panel showing in the browser, which tells you that the certificate is not trusted. You trust it and then it continues. The sense of this action is that the WebApp puts the certificate of the publisher (which is a selfsigned one) into its trusted cert store.
- After this you can try connect to your target OPC UA server. The same procedure will happen with that one.
- Now you should be able to browse your target OPC UA server and also be able to publish node values via right click context menu.

hansgschossmann, 2019-01-29:
I have heard nothing back on this one for a while. Pls reopen if you still see issues.

closed 2019-01-29 by hansgschossmann