Azure/iot-edge-opc-publisher-v1 #151: Receiving Microsoft.Azure.Devices.Client.Exceptions.UnauthorizedException after restarting OpcPublisher.

anhame, 2018-09-05:
Hello,
I'm running the opcpublisher as a native windows app.
When starting it for the first time, everything works fine. After the app has been terminated, subsequent attempts to start it again always results in the following error:

2018-09-05 13:48:59.928 +02:00 [INF] IoTHub device cert store type is: X509Store
2018-09-05 13:48:59.931 +02:00 [INF] IoTHub device cert path is: My
2018-09-05 13:48:59.933 +02:00 [INF] Attempting to register ourselves with IoT Hub using owner connection string.
2018-09-05 13:49:00.851 +02:00 [INF] Device 'IoTDevice' found in IoTHub registry. Remove it.
2018-09-05 13:49:00.932 +02:00 [INF] Adding device 'IoTDevice' to IoTHub registry.
2018-09-05 13:49:01.142 +02:00 [INF] Generate device connection string.
2018-09-05 13:49:01.161 +02:00 [INF] Adding device connectionstring to secure store.
2018-09-05 13:49:01.722 +02:00 [INF] Attempting to read device connection string from cert store using subject name: IoTDevice
2018-09-05 13:49:01.815 +02:00 [INF] IoTCentral mode: False
2018-09-05 13:49:01.823 +02:00 [DBG] Open IoTHub communication
2018-09-05 13:49:02.872 +02:00 [ERR] Failure initializing hub communication processing.
Microsoft.Azure.Devices.Client.Exceptions.UnauthorizedException: CONNECT failed: RefusedNotAuthorized
   at Microsoft.Azure.Devices.Client.InternalClient.<>c.<ApplyTimeout>b__62_2(Task t)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at OpcPublisher.HubCommunication.InitHubCommunicationAsync(TransportType transportType) in C:\Projects\iot-edge-opc-publisher\opcpublisher\HubCommunication.cs:line 152

After some time (approximately an hour), it gets working again.
What could be causing this issue?
best

jamesCBK, 2018-09-05:
Check whether you PC is connected to Internet.

anhame, 2018-09-05:
Thanks, but internet connection is always fine.
I've finally added the device connection string argument within the command line in order to prevent the device always getting recreated. Now it works fine.