Azure/iot-edge-opc-publisher-v1 #155: initial setting owner and/or device connection string as edge module does not work

wmausz, 2018-09-12:
I tried to use iot-edge-opc-publisher on a Linux machine (e.g. https://docs.microsoft.com/en-us/azure/iot-edge/quickstart-linux) using it as module in Azure IoT Edge
When adding the opc-publisher as module, following the documentation shown here, I get errors regarding the owner and device connection string when the module starts.
As edge module the commandline parameter for the owner string is not used. It seems as well that the environment variable _HUB_CS is not used, I have set it in the config page of the module (serveral times)

When starting the solution via docker like shown in your documentation, it seems to work, there I can also use the command line parameter, but as soon as I try it again as edge module, the connection string is not availiable from the command line call before and as well not from other sources like passing the parameter via edge module configuration or environment variable.

Maybe one necessary step for edge modules in the documentation is missing?

from the logs

wmausz, 2018-09-12:
[10:44:46 INF] Current directory is: /appdata
[10:44:46 INF] Log file is: pub-test-publisher.log
[10:44:46 INF] Log level is: info
[10:44:46 INF] Publisher is starting up...
[10:44:46 INF] Application Certificate store type is: X509Store
[10:44:46 INF] Application Certificate store path is: CurrentUser\UA_MachineDefault
[10:44:46 INF] Application Certificate subject name is: publisher
[10:44:46 INF] Application certificate found in Application Certificate Store
[10:44:47 INF] Application certificate is for Application URI 'urn:pub-test:publisher:microsoft:', Application 'publisher and has Subject 'publisher'
[10:44:47 INF] Trusted Issuer store type is: Directory
[10:44:47 INF] Trusted Issuer Certificate store path is: CertificateStores/issuers
[10:44:47 INF] Trusted Peer Certificate store type is: Directory
[10:44:47 INF] Trusted Peer Certificate store path is: CertificateStores/trusted
[10:44:47 INF] Rejected certificate store type is: Directory
[10:44:47 INF] Rejected Certificate store path is: CertificateStores/rejected
[10:44:47 INF] Rejection of SHA1 signed certificates is disabled
[10:44:47 INF] Minimum certificate key size set to 1024
[10:44:47 INF] Adding publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
[10:44:47 INF] A certificate with the same thumbprint is already in the trusted store.
[10:44:47 INF] OperationTimeout set to 120000
[10:44:47 INF] Publisher server base address: opc.tcp://pub-test:62222/UA/Publisher
[10:44:47 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[10:44:47 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
[10:44:47 INF] opcstacktracemask set to: 0x0
[10:44:47 INF] There is no site configured.
[10:44:47 INF] Publisher configured to auto trust server certificates of the servers it is connecting to.
[10:44:47 INF] Starting server on endpoint opc.tcp://pub-test:62222/UA/Publisher ...
[10:44:52 INF] Server started.
[10:44:52 INF] Using default telemetry configuration.
[10:44:52 INF] The name of the configuration file for published nodes is: ./pn.json
[10:44:52 INF] The node configuration file './pn.json' does not exist. Continue and wait for remote configuration requests.
[10:44:52 INF] There are 0 nodes to publish.
[10:44:52 INF] IoT Hub owner connection string not passed as argument.
[10:44:52 INF] IoTHub device cert store type is: X509Store
[10:44:52 INF] IoTHub device cert path is: My
[10:44:52 INF] IoT Hub owner connection string not specified. Assume device connection string already in cert store or passed in via command line option.
[10:44:52 INF] Attempting to read device connection string from cert store using subject name: publisher
[10:44:52 FTL] Device connection string not found in secure store. Please pass it in at least once via command line option. Can not connect to IoTHub. Exiting..

payalgaikwad42, 2018-10-16:
Same thing is happening with me, i am deploying OPC publisher module as an  iot edge module but it is not getting deployed. 
@wmausz can you tell me steps which you have followed for deployment ?

wmausz, 2018-10-17:
Hello! It is missing in the documentation somehow - one should clarify
that.
I started the OPC publisher one time as "pure" container via command line:
"docker run microsoft/iot-edge-opc-publisher <applicationname>
[<iothubconnectionstring>] [options]"
You have to specify excatly the application name that you will use in the
configuration of the edge module - "publisher" - this will than save the
connection to the internal store and than the edge module can get it from
there.
How to set it only via edge module without connecting to the device -
actually no idea until now. The problem is when I pass the connection
string as parameter in the edge module configuration, I get the error that
the configuration string is too long. So passing it as container config
will not work.

hansgschossmann, 2018-11-07:
The documentation is correct. When you run OPC Publisher as an IoT Edge module, it sends data to IoTHub via EdgeHub (one of the two system modules of IoT Edge). The connection string to connect to EdgeHub is build at runtime. So no need to configure OPC Publisher with it.

But you need to configure IoT Edge with the device connection string of the edge device identity as described here: https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-linux and the pages for different other platforms and host systems.

Please reopen the issue, if you still see issues.

closed 2018-11-07 by hansgschossmann