Azure/iot-edge-opc-publisher-v1 #159: Custom Create Options cannot exceed 512 bytes

mawaak, 2018-10-31:
Hi,

When following the guide to set up the module on the IoT edge I am blocked by following error:

> Custom Create Options cannot exceed 512 bytes

When trying to use the specified Container Create Options:

 ```
 {
      "Hostname": "publixher",
      "Cmd": [
          "publisher",
          "--pf=./pn.json",
          "--di=60",
          "--to",
          "--aa",
          "--si=0",
          "--ms=0"
      ],
      "HostConfig": {
          "PortBindings": {
              "62222/tcp": [{
                  "HostPort": "62222"
              }]
          },
          "Binds": [
              "d:/iiotedge:/appdata"
          ],
          "ExtraHosts": [
              "localhost:127.0.0.1",
              "opctestsvr:192.168.178.26"
          ]
      }
  }
```

Any suggestion to how to get around this?

sandervandevelde, 2018-11-06:
I had the same issue. With the recent 1.0.4 release of IoT Edge runtime/modules this should be fixed (size is [now limited to 4K](https://azure.microsoft.com/en-us/updates/iot-edge1-0-4/)).

hansgschossmann, 2018-11-07:
@sandervandevelde i just checked in the Azure portal and it is still showing the limit of 512. I keep this one open. I have already followed up with the portal folks and asked if they can disregard the whitespaces in the check. I will also check if/when they will support the 4k size.

hansgschossmann, 2019-01-29:
Allright, i just checked it is now accepting Container Create Options in the portal up to 4096 byte.

closed 2019-01-29 by hansgschossmann