Azure/iot-edge-opc-publisher-v1 #160: value is empty

dcrupel, 2018-10-31:
Hi,

The value of my server is always empty. Anyone who has had this problem?

What I receive in the opc publisher module (value null);
2018-10-31 11:58:44.298 +00:00 [VRB] OPC: PUBLISH #5 RECEIVED
2018-10-31 11:58:44.299 +00:00 [VRB] OPC: Publish Completed. RequestHandle=5, PendingRequestCount=1
2018-10-31 11:58:44.300 +00:00 [VRB] OPC: NOTIFICATION RECEIVED: SubId=20, SeqNo=1
2018-10-31 11:58:44.301 +00:00 [VRB] OPC: Publish Called. RequestHandle=7, PendingRequestCount=2
2018-10-31 11:58:44.302 +00:00 [VRB] OPC: PUBLISH #7 SENT
2018-10-31 11:58:44.304 +00:00 [VRB] OPC: NotificationReceived: ClientHandle=219, Value=
2018-10-31 11:58:44.308 +00:00 [DBG]    ApplicationUri: urn:ti-fabb442c:UA:ComServerWrapper
2018-10-31 11:58:44.308 +00:00 [DBG]    EndpointUrl: opc.tcp://192.168.31.29:48400/UA/ComServerWrapper
2018-10-31 11:58:44.309 +00:00 [DBG]    DisplayName: ns=2;s=0:OPC-Tick
2018-10-31 11:58:44.309 +00:00 [DBG]    Value: 

If I check with an OPC UA Client I always get a value:
[image]

Thanks

Dries

hansgschossmann, 2018-11-07:
Can you run OPC Publisher under Visual Studio and check which value is reported in monitoredItem of MonitoredItem_Notification?

dcrupel, 2018-11-28:
It looks like OPC DA Server 1 takes to long to connect to which causes that the publisher module returns a null value and StatusCode {BadWaitingForInitialData}. Is there a way to make the publisher tries until he gets a value from the server? (both servers are wrapped into 1 OPC UA Server on which the publisher is listening).

OPC DA Server 1:
[image]

OPC DA Server 2:
[image]

OPC-Tick = node on OPC DA Server 1
increment.UI1 = node on OPC DA Server 2
[image]

Thanks!

Regards