Azure/iot-edge-opc-publisher-v1 #167: Hub connection fails with certificate validation error

Strandfelt, 2018-11-22:
As of today, I am having problems with deploying the OPC Publisher edge module.

I am getting the following error:
```
[09:54:19 ERR] Failure initializing hub communication processing.
System.Security.Authentication.AuthenticationException: The remote certificate is invalid according to the validation procedure.
   at Microsoft.Azure.Devices.Client.InternalClient.<>c.<ApplyTimeout>b__62_2(Task t)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at OpcPublisher.HubCommunication.InitHubCommunicationAsync(TransportType transportType) in /app/opcpublisher/HubCommunication.cs:line 128
```

**Deployment information:**
Host machine: Windows 10
Image URI: `microsoft/iot-edge-opc-publisher:latest`
Docker VM: Linux

Container create options:
```
{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=./pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=0",
    "--ms=0",
    "--fd=True"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "x509certstores:/root/.dotnet/corefx/cryptography/x509stores",
      "d:/iotedge:/appdata"
    ],
    "ExtraHosts": [
      "localhost:127.0.0.1",
      "opcserverhostname:192.168.1.XXX"
    ]
  }
}
```

OPC Publisher log content:
```
2018-11-22 09:35:55.481 +00:00 [INF] Current directory is: /appdata
2018-11-22 09:35:55.592 +00:00 [INF] Log file is: publisher-publisher.log
2018-11-22 09:35:55.592 +00:00 [INF] Log level is: info
2018-11-22 09:35:55.592 +00:00 [INF] Publisher is starting up...
2018-11-22 09:35:55.611 +00:00 [INF] Application Certificate store type is: X509Store
2018-11-22 09:35:55.612 +00:00 [INF] Application Certificate store path is: CurrentUser\UA_MachineDefault
2018-11-22 09:35:55.612 +00:00 [INF] Application Certificate subject name is: publisher
2018-11-22 09:35:55.649 +00:00 [INF] Application certificate found in Application Certificate Store
2018-11-22 09:35:55.678 +00:00 [INF] Application certificate is for Application URI 'urn:publisher:publisher:microsoft:', Application 'publisher and has Subject 'publisher'
2018-11-22 09:35:55.679 +00:00 [INF] Trusted Issuer store type is: Directory
2018-11-22 09:35:55.679 +00:00 [INF] Trusted Issuer Certificate store path is: CertificateStores/issuers
2018-11-22 09:35:55.679 +00:00 [INF] Trusted Peer Certificate store type is: Directory
2018-11-22 09:35:55.679 +00:00 [INF] Trusted Peer Certificate store path is: CertificateStores/trusted
2018-11-22 09:35:55.679 +00:00 [INF] Rejected certificate store type is: Directory
2018-11-22 09:35:55.679 +00:00 [INF] Rejected Certificate store path is: CertificateStores/rejected
2018-11-22 09:35:55.679 +00:00 [INF] Rejection of SHA1 signed certificates is disabled
2018-11-22 09:35:55.679 +00:00 [INF] Minimum certificate key size set to 1024
2018-11-22 09:35:55.683 +00:00 [INF] Adding publisher certificate to trusted peer store. StorePath=CertificateStores/trusted
2018-11-22 09:35:55.709 +00:00 [INF] OperationTimeout set to 120000
2018-11-22 09:35:55.711 +00:00 [INF] Publisher server base address: opc.tcp://publisher:62222/UA/Publisher
2018-11-22 09:35:55.712 +00:00 [INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
2018-11-22 09:35:55.712 +00:00 [INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
2018-11-22 09:35:55.714 +00:00 [INF] opcstacktracemask set to: 0x0
2018-11-22 09:35:55.727 +00:00 [INF] There is no site configured.
2018-11-22 09:35:55.727 +00:00 [INF] Publisher configured to auto trust server certificates of the servers it is connecting to.
2018-11-22 09:35:55.728 +00:00 [INF] Starting server on endpoint opc.tcp://publisher:62222/UA/Publisher ...
2018-11-22 09:35:56.643 +00:00 [INF] Server started.
2018-11-22 09:35:56.655 +00:00 [INF] Using default telemetry configuration.
2018-11-22 09:35:56.671 +00:00 [INF] The name of the configuration file for published nodes is: ./pn.json
2018-11-22 09:35:56.672 +00:00 [INF] Attemtping to load node configuration from: ./pn.json
2018-11-22 09:35:56.873 +00:00 [INF] Loaded 2 config file entry/entries.
2018-11-22 09:35:56.887 +00:00 [INF] There are 7568 nodes to publish.
2018-11-22 09:35:56.890 +00:00 [INF] Create IoTEdgeHub module client using 'Amqp_Tcp_Only' for communication.
2018-11-22 09:35:57.745 +00:00 [INF] IoTCentral mode: False
2018-11-22 09:35:58.284 +00:00 [ERR] Failure initializing hub communication processing.
System.Security.Authentication.AuthenticationException: The remote certificate is invalid according to the validation procedure.
   at Microsoft.Azure.Devices.Client.InternalClient.<>c.<ApplyTimeout>b__62_2(Task t)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at OpcPublisher.HubCommunication.InitHubCommunicationAsync(TransportType transportType) in /app/opcpublisher/HubCommunication.cs:line 128
```

Strandfelt, 2018-11-22:
I removed the `"x509certstores:/root/.dotnet/corefx/cryptography/x509stores",` part from the container create options and that seems to do the trick. Not sure why, but I see that the line has also been removed in the documentation of the module. I am pretty sure I got it from the documentation on this git previously.

hansgschossmann, 2019-01-29:
I have updated the documentation shortly after you hit the issue. IoT Edge had mapped the same directory in their GA version, which then caused an issue when OPC Publisher mapped it away.

closed 2019-01-29 by hansgschossmann