Azure/iot-edge-opc-publisher-v1 #176: GetPublishedNodes with error 'Bad' with publisher 2.3.0

zhironghsu, 2019-01-16:
Hi,

I want to get all nodes of all endpoints by calling 'GetPublishedNodes' method without input EndpointUrl , and it works before the version 2.3.0.

But I got an error 'Bad' on the publisher 2.3.0.

Any suggestions?
Thank you!

zhironghsu, 2019-01-18:
Is there an OPC UA method same as `GetConfiguredEndpoints`?  
My project is based on OPC UA methods, and my current approach is to call `GetPublishedNodes` without input EndpointUrl to get all the configured endpoints and then call `GetPublishedNodes` with EndpointUrl to get configured nodes on the endpoint. It seems I need to move to IoT Hub methods in the future. But for backward compatibility (for the user using the publisher version which has no IoT Hub methods), it would be better to keep the legacy OPC UA method calls.

hansgschossmann, 2019-01-18:
Oh, sorry. I am so focused on the IoT Hub direct methods. I see your need now. 
I have stopped supporting empty input parameters, since the returned value could exceed the size of the OPC msg and causing an exception in the method. 
Now can i reenable it and deliver what fits in, but this is not nice.
Coincidentally @marcschier has created a PR today which is wiring through an OPC method call to the IoTHub direct method call. Can you take a look at the PR here: https://github.com/Azure/iot-edge-opc-publisher/pull/179

I have asked him to add the same for GetConfiguredEndpoints. Lets see if he can make it otherwise you or i will do it and then you should be unblocked. ETA should be mid/end next week.

zhironghsu, 2019-01-21:
I've tried the new `GetConfiguredEndpoints` OPC method, and it works as expected. 
Thank you.

When will it have an official release (Docker image) for this version?

hansgschossmann, 2019-01-21:
Currently the functionality is available as latest. Please do a bit more of testing and I release it as 2.3.1, before I merge in more changes.

hansgschossmann, 2019-01-23:
Hi @zhironghsu, 
as indicated earlier it will take this week to get this released. After some further thinking, i removed the two new OPC methods again and will add a generic OPC method to indirectly call the IoTHub direct methods. Pls check out branch develop_hans_opcmethod, the Linux container is available as hansgschossmann/iot-edge-opc-publisher:om or for windows as hansgschossmann/iot-edge-opc-publisher:om-windows-amd64
The first string parameter of the OPC method IoTHubDirectMethod is the method name, the second one the JSON payload as it is for the IoTHub direct method.

Can you pls give it a try?

zhironghsu, 2019-01-24:
Hi @hansgschossmann,

It's a good idea to have a generic OPC method to wire all the IoTHub direct methods. 
I've tried the `IoTHubDirectMethod`  with the method name `GetConfiguredEndpoints`, `GetConfiguredNodesOnEndpoint` and it works.

I found that if I input an incorrect method name as the **MethodName** parameter of `IoTHubDirectMethod`, it returns a `Good` status.

How about to return an OPC Bad status code such like `Opc.Ua.StatusCodes.BadInvalidArgument`, 
`Opc.Ua.StatusCodes.BadMethodInvalid`, or `Opc.Ua.StatusCodes.BadNotImplemented`  to make the result more clear?

Thanks.

hansgschossmann, 2019-01-27:
I have released 2.3.1 today. Pls reopen if you still see issues.

closed 2019-01-27 by hansgschossmann