Azure/iot-edge-opc-publisher-v1 #177: Direct method PublishNodes node configuration is useless after UnpublishAllNodes

sandervandevelde, 2019-01-18:
If I execute 

    var _method = new CloudToDeviceMethod("UnpublishAllNodes", responseTimeout, connectionTimeout);

And try to add a new configuration using 'PublishNodes', the "UseSecurity" (I set it to False) element is ignored. Therefore the connection to the device is rejected.

I have to repair the json file with the configuration by hand to fix it.

hansgschossmann, 2019-01-18:
hi sander, yes that is an bug. thanks for opening the issue.

i think the fix is simple. in the HandlePublishNodesMethodAsync method this line:
                            opcSession = new OpcSession(endpointUri.OriginalString, true, OpcSessionCreationTimeout);

is passing in true as useSecurity into the ctor instead of what is passed in via the method.

Do you want to create a PR? If you have no time i will change it somewhen.

sandervandevelde, 2019-01-30:
Hello Hans, the pull request https://github.com/Azure/iot-edge-opc-publisher/pull/184 is created for this issue.

hansgschossmann, 2019-02-01:
@sandervandevelde  yes this is by design. The PublishNode OPC method does not accept a parameter to set the security for an endpoint. The new OPC method to call IoTHub direct methods allows to initiate a session with an unsecure transport via OPC if needed.

closed 2019-02-01 by hansgschossmann