Azure/iot-edge-opc-publisher-v1 #181: Issue in Building Windows Docker Image

Engr1979, 2019-01-18:
When building OPC Publisher container using below command

docker build -f <docker-configfile-to-use> -t <your-container-name> .

I get an error in **Step 7/16 : RUN dotnet restore**
The error says:
**C:\Program Files\dotnet\sdk\2.1.503\NuGet.targets(114,5): error : Unable to load service index for source https://api.nuget.org/v3/index.json.**

The system I am building is a clean Win10 Ent LTSC edition with just the Windows SDK installed.
Quite a few solutions on web are pointing to correcting the proxy settings in env variables which I dont have in my system. Also, some solutions are suggesting to modify NuGet config file. But, I dont have any NuGet packages in my system as it is a clean OS. I am for sure not looking to install any version of VS on my system.

Is it a pre-requisite to install any NuGet Package before the docker container be built?

Any suggestions on how to move ahead from here?

hansgschossmann, 2019-01-22:
I never saw that. docker build does not require any installations on the host. I would try to add "-v detailed" to the "RUN dotnet restore" to see if if shows any more details. 
Since it sounds like you can not access the resource you may try:
docker run -it busybox
And in the busybox command prompt:
wget https://api.nuget.org/v3/index.json

Engr1979, 2019-01-28:
Thanks for the Inputs. Although I was having internet connectivity it was unstable causing the issue. After I got a stable connection, it worked fine. Thanks for directing me to probable cause.

closed 2019-01-28 by Engr1979