Azure/iot-edge-opc-publisher-v1 #186: Invalid Volume Specification when using Windows container on Windows

sandervandevelde, 2019-02-04:
On https://hub.docker.com/_/microsoft-iotedge-opc-publisher I find the latest release of the Windows container for deployment on Windows IoT Enterprise 1809.
I use the 1.0.6 runtime deployment which uses Moby. Other modules deploy well and run as expected.

But the OPCPublisher crashes on creation with message ...

    Could not create module opcPublisher 
    caused by: invalid volume specification: 'c:/iiotedge:/appdata', StatusCode 500 

... when using the create options:

> {
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=./pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=10",
    "--ms=262144"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "c:/iiotedge:/appdata"
    ],
    "ExtraHosts": [
      "localhost:127.0.0.1"
    ]
  }
}

One possible difference regarding Docker and Moby is that Docker needs access permissions to the C drive. 
Is this also the case for Moby? If so, how?
Or am I just misspelling the path?

Strandfelt, 2019-02-07:
We experienced the same issue when switching from Linux Containers to Windows Containers. We solved it by changing the Bind value to Capital drive letters: `"D:/iotedge:C:/appdata"`
I don't know if it's the right thing to do according to the documentation, but it works.

Full example: 
```
{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=./pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=0",
    "--ms=0",
    "--fd=True"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "D:/iotedge:C:/appdata"
    ],
    "ExtraHosts": [

    ]
  }
}
```

hansgschossmann, 2019-02-07:
it is the right thing to do. on windows the default working directory is not "/appdata", but "C:/appdata", i keep this one open to update the documentation.

sandervandevelde, 2019-02-07:
@Strandfelt Thanks for pointing me in the right direction!

I got it working with 'c:/iotedge:c:/appdata'.

Yes, it works with (on my machine :-) Win10IoT Enterprise 1809 with iot edge runtime 1.0.6) both with capital and non-capital drive letters.

**It seems the 'c:/appdata' folder is required!**

I tried several other container folder names but only 'c:/appdata' resulted in the pn.json being found.

hansgschossmann, 2019-02-07:
the -v option specifies the <folder on host>:<folder in container>, the [docker build script](https://github.com/Azure/iot-edge-opc-publisher/blob/master/docker/windows/amd64/Dockerfile) set the working directory to "/appdata", so this needs to be the <folder in container> and on Windows it will be on C:. I have not tried what happens if you use --pf to get the config file from somewhere else.
@Strandfelt are you saying that if the c:/appdata is not there also you have mapped it to somewhere else via -v the startup fails?

sandervandevelde, 2019-02-07:
I have not taken the -v parameter into account. I only set these container create options:

> {
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=./pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=10",
    "--ms=262144",
    "--fd=True"
  ],
  "HostConfig": {
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "Binds": [
      "c:/iotedge:c:/appdata"
    ],
    "ExtraHosts": []
  }
}

I just wanted to confirm the 'c:/appdata' with or without capital drive letter.

I have seen a more flexible approach. Look at a simular situation in the edgeHub for [offline storage](https://docs.microsoft.com/en-us/azure/iot-edge/offline-capabilities). The bind is combined with an environment variable.

hansgschossmann, 2019-02-07:
The -v option is if you start docker run via command line and is the equivalent to the Binds in the container create options. If you like the environment approach more, then you can maybe use the _GW_PNFP env variable which is used by OPC Publisher to read the file name of the configuration file from if it is set.
@sandervandevelde so my question is still: are you saying C:\appdata must be there even you are using a different host path in the Binds option?

sandervandevelde, 2019-02-07:
No, only c:\iotedge needs to be created on the host operating system, I have no c:\appdata created on the host C drive.

hansgschossmann, 2019-02-13:
i added some more documentation to the readme.

closed 2019-02-13 by hansgschossmann