Azure/iot-edge-opc-publisher-v1 #188: OPC Publisher not reconnecting after edgeHub crash

BewaControl-ReneDivossen, 2019-02-08:
After edgeHub crashes and recovers, the OPC Publisher looses it's connection to edgeHub.  No Messages are received in edgeHub anymore.

We route the messages send by OPC Publisher through some custom modules which generate messages to be send to IotHub.  After a crash of edgeHub the custom modules do not receive messages from OPC Publisher anymore.  After "sudo docker stop publisher"  the OPC Publisher module recovers and starts sending messages again.

BewaControl-ReneDivossen, 2019-02-12:
With the latest release and using AMQP it's just the other way around. 
If I kill the edgeHub module using "docker stop" everything will recover nicely but if I kill the publisher module using "docker stop" it will not send messages anymore after coming up running again.

BewaControl-ReneDivossen, 2019-02-13:
For AMQP everything is working after this [commit](https://github.com/Azure/iot-edge-opc-publisher/commit/8f1d85f284bc9891b97eb7e683ad00ed3816072f)

hansgschossmann, 2019-02-13:
fixed with 2.3.3

closed 2019-02-13 by hansgschossmann