Azure/iot-edge-opc-publisher-v1 #189: Amqp does not work with IoTEdge

BewaControl-ReneDivossen, 2019-02-08:
I'm using OPC Publisher as IoTEdge module. In the create options I defined the following command-line options:
```
"Cmd": [
                  "publisher",
                  "--pf=./pn.json",
                  "--tc=./telemetryConfig.json",
                  "--di=60",
                  "--to",
                  "--aa",
                  "--si=0",
                  "--ms=0",
                  "--ih=Amqp_Tcp_Only"
                ],
```
I defined to use the Amqp Protocoll (which is also used by all my other modules).
But OpcPublisher is connecting using the MQTT-Protocoll.

In the logging output of the publisher module I found the following line:
_When running as IoTEdge module Mqtt_Tcp_Only is enforced._
Why is that?

sebader, 2019-02-08:
Fixed in https://github.com/Azure/iot-edge-opc-publisher/pull/191

BewaControl-ReneDivossen, 2019-02-08:
@sebader 
Will this be a new release (2.3.2) ? Will there be a docker image for that?

sebader, 2019-02-08:
I'll let @hansgschossmann comment on this but yes, I expect there will be a new release once the PR is merged

BewaControl-ReneDivossen, 2019-02-08:
@sebader 
Thank's a lot for the quick reaction.
@hansgschossmann : It would be great to have a new release soon using the device sdk 1.19.0 and allowing to use AMQP with IotEdge.  This would hopefully allow for lossless and automatic recovery after an EdgeHub crash (which unfortunately right now seems to be inevitable because edgehub consumes more and more memory with time and sooner or later will crash)

BewaControl-ReneDivossen, 2019-02-11:
@hansgschossmann I referenced opc-publisher:latest and I'm using Amqp as the only protocol now. It's running but automatic recovery does not work as expected. If I kill the edgeHub module using "docker stop" everything will recover nicely but if I kill the publisher module using "docker stop" it will not send messages anymore after coming up running again.

hansgschossmann, 2019-02-11:
Can you share the log output of publisher?

Outlook for Android<https://aka.ms/ghei36> herunterladen

BewaControl-ReneDivossen, 2019-02-11:
@hansgschossmann Here comes the log output (Starting with the Publisher running and sending messages.  Then the Publisher docker image has been stopped and after recovering does not send messages anymore.  Then EdgeHub docker image has been stopped and after recovering Publisher send messages again):

`
[08:44:37 INF] ==========================================================================
[08:44:37 INF] OpcPublisher status @ 02/11/2019 08:44:37 (started @ 02/11/2019 08:38:18)
[08:44:37 INF] ---------------------------------
[08:44:37 INF] OPC sessions (configured/connected): 1/1
[08:44:37 INF] OPC subscriptions (configured/connected): 1/1
[08:44:37 INF] OPC monitored items (configured/monitored/to remove): 4/4/0
[08:44:37 INF] ---------------------------------
[08:44:37 INF] monitored items queue bounded capacity: 8192
[08:44:37 INF] monitored items queue current items: 0
[08:44:37 INF] monitored item notifications enqueued: 403
[08:44:37 INF] monitored item notifications enqueue failure: 0
[08:44:37 INF] ---------------------------------
[08:44:37 INF] messages sent to IoTHub: 402
[08:44:37 INF] last successful msg sent @: 02/11/2019 08:44:35
[08:44:37 INF] bytes sent to IoTHub: 23837
[08:44:37 INF] avg msg size: 59
[08:44:37 INF] msg send failures: 1
[08:44:37 INF] messages too large to sent to IoTHub: 0
[08:44:37 INF] times we missed send interval: 0
[08:44:37 INF] number of events: 403
[08:44:37 INF] ---------------------------------
[08:44:37 INF] current working set in MB: 76
[08:44:37 INF] --si setting: 0
[08:44:37 INF] --ms setting: 0
[08:44:37 INF] --ih setting: Amqp_Tcp_Only
[08:44:37 INF] ==========================================================================
[08:45:37 INF] ==========================================================================
[08:45:37 INF] OpcPublisher status @ 02/11/2019 08:45:37 (started @ 02/11/2019 08:38:18)
[08:45:37 INF] ---------------------------------
[08:45:37 INF] OPC sessions (configured/connected): 1/1
[08:45:37 INF] OPC subscriptions (configured/connected): 1/1
[08:45:37 INF] OPC monitored items (configured/monitored/to remove): 4/4/0
[08:45:37 INF] ---------------------------------
[08:45:37 INF] monitored items queue bounded capacity: 8192
[08:45:37 INF] monitored items queue current items: 1
[08:45:37 INF] monitored item notifications enqueued: 473
[08:45:37 INF] monitored item notifications enqueue failure: 0
[08:45:37 INF] ---------------------------------
[08:45:37 INF] messages sent to IoTHub: 470
[08:45:37 INF] last successful msg sent @: 02/11/2019 08:45:35
[08:45:37 INF] bytes sent to IoTHub: 27913
[08:45:37 INF] avg msg size: 59
[08:45:37 INF] msg send failures: 1
[08:45:37 INF] messages too large to sent to IoTHub: 0
[08:45:37 INF] times we missed send interval: 0
[08:45:37 INF] number of events: 472
[08:45:37 INF] ---------------------------------
[08:45:37 INF] current working set in MB: 76
[08:45:37 INF] --si setting: 0
[08:45:37 INF] --ms setting: 0
[08:45:37 INF] --ih setting: Amqp_Tcp_Only
[08:45:37 INF] ==========================================================================
IoTEdge detected.
[08:46:48 INF] Current directory is: /appdata
[08:46:48 INF] Log file is: publisher-publisher.log
[08:46:48 INF] Log level is: info
[08:46:48 INF] OPC Publisher V2.3.2+x starting up...
[08:46:48 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[08:46:48 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[08:46:48 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[08:46:49 INF] opcstacktracemask set to: 0x0
[08:46:49 INF] OperationTimeout set to 120000
[08:46:49 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[08:46:49 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[08:46:49 INF] Trusted Issuer store type is: Directory
[08:46:49 INF] Trusted Issuer Certificate store path is: pki/issuer
[08:46:49 INF] Trusted Peer Certificate store type is: Directory
[... 133 lines cut ...]
[08:51:08 INF] OpcPublisher status @ 02/11/2019 08:51:08 (started @ 02/11/2019 08:46:47)
[08:51:08 INF] ---------------------------------
[08:51:08 INF] OPC sessions (configured/connected): 1/1
[08:51:08 INF] OPC subscriptions (configured/connected): 1/1
[08:51:08 INF] OPC monitored items (configured/monitored/to remove): 4/4/0
[08:51:08 INF] ---------------------------------
[08:51:08 INF] monitored items queue bounded capacity: 8192
[08:51:08 INF] monitored items queue current items: 0
[08:51:08 INF] monitored item notifications enqueued: 269
[08:51:08 INF] monitored item notifications enqueue failure: 0
[08:51:08 INF] ---------------------------------
[08:51:08 INF] messages sent to IoTHub: 269
[08:51:08 INF] last successful msg sent @: 02/11/2019 08:51:07
[08:51:08 INF] bytes sent to IoTHub: 15911
[08:51:08 INF] avg msg size: 59
[08:51:08 INF] msg send failures: 0
[08:51:08 INF] messages too large to sent to IoTHub: 0
[08:51:08 INF] times we missed send interval: 0
[08:51:08 INF] number of events: 269
[08:51:08 INF] ---------------------------------
[08:51:08 INF] current working set in MB: 77
[08:51:08 INF] --si setting: 0
[08:51:08 INF] --ms setting: 0
[08:51:08 INF] --ih setting: Amqp_Tcp_Only
[08:51:08 INF] ==========================================================================`

dmakarenko-connyun, 2019-02-11:
I built am image from the master branch and C2D message doesn't work with AMQP protocol (which is default now). Setting --ih=Mqtt_Tcp_Only fixes the problem.

hansgschossmann, 2019-02-12:
I have looked into the Amqp issue today, but was not able to root cause the issue. If i revert Microsoft.Azure.Devices.Client back to 1.17.1 then C2D methods using Amqp starts working. Need to dig deeper.

hansgschossmann, 2019-02-13:
@dmakarenko-connyun, @BewaControl-ReneDivossen: i have now pushed a fix which should enable Amqp as transport when running as IoT Edge module. If you have a chance, pls give it a try.

dmakarenko-connyun, 2019-02-13:
@hansgschossmann Works for me. Thanks!

BewaControl-ReneDivossen, 2019-02-13:
@hansgschossmann  My problem has also been solved with this fix.  

OpcPublisher recovers after all of the following:
- docker stop publisher
- docker stop edgeHub
- systemctl restart iotege
- reboot

Thank you very much for the quick response!

hansgschossmann, 2019-02-13:
@BewaControl-ReneDivossen this is good to hear, i just have tested this as well and see no issues. i will kick of a tagged build today.

hansgschossmann, 2019-02-13:
@BewaControl-ReneDivossen now the 2.3.3 release is available. Thanks for all your feedback and testing.

closed 2019-02-13 by hansgschossmann

dmakarenko-connyun, 2019-02-14:
@hansgschossmann today morning my publisher module was not responsive again. I was not able to send C2D messages for publisher but my own module connected using MQTT was very OK (both modules were running since yesterday).
Restarting the module didn't help, only after restarting the iotedge itself (`sudo systemctl restart iotedge`) the module was able to receive messages. 
Unfortunately I can't provide publisher logs and will investigate it further but using AMQP in production looks for not so reliable because of [this](https://github.com/Azure/azure-iot-sdk-csharp/issues/558).

BewaControl-ReneDivossen, 2019-02-14:
@dmakarenko-connyun The last comments on the linked issue states that reconnection of modules works for AMQP but not for MQTT.  Did you try using AMQP for all your modules and disabling the MQTT head?

dmakarenko-connyun, 2019-02-14:
@BewaControl-ReneDivossen sorry, my fault, looks like the similar issue for AMQP was [fixed](https://github.com/Azure/azure-iot-sdk-csharp/issues/571).