Azure/iot-edge-opc-publisher-v1 #195: TLS/SSL connection of Publisher to IoT-hub fails

JuergenKosel, 2019-02-13:
Hello,

I try to update the OPC-Publisher on an embedded Linux (arm) device to the tagged version 2.3.2 .
As the log below indicates, the connection of the publisher to the IoT-Hub fails, due to one missing file.
Unfortunally the log message does not say the exact name of the missing file.

```
2019-02-13 13:43:15.895 +00:00 [INF] Current directory is: /var/www/Azure
2019-02-13 13:43:16.548 +00:00 [INF] Log file is: /var/tmp/publisher.txt
2019-02-13 13:43:16.549 +00:00 [INF] Log level is: info
2019-02-13 13:43:16.659 +00:00 [INF] OPC Publisher V2.3.0 starting up...
2019-02-13 13:43:16.678 +00:00 [INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
2019-02-13 13:43:16.682 +00:00 [INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
2019-02-13 13:43:16.682 +00:00 [INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
2019-02-13 13:43:16.844 +00:00 [INF] opcstacktracemask set to: 0x0
2019-02-13 13:43:16.852 +00:00 [INF] OperationTimeout set to 120000
2019-02-13 13:43:16.876 +00:00 [INF] OPC UA server base address: opc.tcp://edgegate00f5:62222/UA/Publisher
2019-02-13 13:43:16.883 +00:00 [INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
2019-02-13 13:43:17.075 +00:00 [INF] Trusted Issuer store type is: Directory
2019-02-13 13:43:17.078 +00:00 [INF] Trusted Issuer Certificate store path is: pki/issuer
2019-02-13 13:43:17.080 +00:00 [INF] Trusted Peer Certificate store type is: Directory
2019-02-13 13:43:17.080 +00:00 [INF] Trusted Peer Certificate store path is: pki/trusted
2019-02-13 13:43:17.082 +00:00 [INF] Rejected certificate store type is: Directory
2019-02-13 13:43:17.082 +00:00 [INF] Rejected Certificate store path is: pki/rejected
2019-02-13 13:43:17.084 +00:00 [INF] Rejection of SHA1 signed certificates is disabled
2019-02-13 13:43:17.086 +00:00 [INF] Minimum certificate key size set to 1024
2019-02-13 13:43:17.094 +00:00 [INF] Application Certificate store type is: Directory
2019-02-13 13:43:17.097 +00:00 [INF] Application Certificate store path is: pki/own
2019-02-13 13:43:17.098 +00:00 [INF] Application Certificate subject name is: edgegate00F5
2019-02-13 13:43:17.099 +00:00 [WRN] WARNING: Automatically accepting certificates. This is a security risk.
2019-02-13 13:43:17.425 +00:00 [INF] Application certificate with thumbprint 'CC819F8673D96091825A5FC9FED09BAA151F3155' found in the application certificate store.
2019-02-13 13:43:17.674 +00:00 [INF] Application certificate is for ApplicationUri 'urn:edgegate00f5:edgegate00F5.softing-testnet-dmz.org:microsoft:', ApplicationName 'edgegate00F5' and Subject is 'edgegate00F5'
2019-02-13 13:43:17.676 +00:00 [INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
2019-02-13 13:43:17.776 +00:00 [INF] Trusted issuer store contains 0 certs
2019-02-13 13:43:17.787 +00:00 [INF] Trusted issuer store has 0 CRLs.
2019-02-13 13:43:17.789 +00:00 [INF] Trusted peer store contains 0 certs
2019-02-13 13:43:17.790 +00:00 [INF] Trusted peer store has 0 CRLs.
2019-02-13 13:43:17.790 +00:00 [INF] Rejected certificate store contains 0 certs
2019-02-13 13:43:17.792 +00:00 [INF] There is no site configured.
2019-02-13 13:43:17.794 +00:00 [INF] Starting server on endpoint opc.tcp://edgegate00f5:62222/UA/Publisher ...
2019-02-13 13:43:32.117 +00:00 [INF] Server started.
2019-02-13 13:43:32.205 +00:00 [INF] Using default telemetry configuration.
2019-02-13 13:43:32.304 +00:00 [INF] The name of the configuration file for published nodes is: /var/www/Azure/publishednodes.json
2019-02-13 13:43:32.305 +00:00 [INF] The node configuration file '/var/www/Azure/publishednodes.json' does not exist. Continue and wait for remote configuration requests.
2019-02-13 13:43:32.305 +00:00 [INF] There are 0 nodes to publish.
2019-02-13 13:43:32.421 +00:00 [INF] IoTHub device cert store type is: X509Store
2019-02-13 13:43:32.422 +00:00 [INF] IoTHub device cert path is: My
2019-02-13 13:43:32.422 +00:00 [INF] Attempting to register ourselves with IoT Hub using owner connection string.
2019-02-13 13:43:34.975 +00:00 [FTL] Error in IoTHub initialization.
Microsoft.Azure.Devices.Common.Exceptions.IotHubCommunicationException: The SSL connection could not be established, see inner exception. ---> System.Net.Http.HttpRequestException: The SSL connection could not be established, see inner exception. ---> Interop+Crypto+OpenSslCryptographicException: error:2006D080:BIO routines:BIO_new_file:no such file
   at Interop.Crypto.CheckValidOpenSslHandle(SafeHandle handle)
   at Internal.Cryptography.Pal.StorePal.LoadMachineStores()
   at Internal.Cryptography.Pal.StorePal.FromSystemStore(String storeName, StoreLocation storeLocation, OpenFlags openFlags)
   at System.Security.Cryptography.X509Certificates.X509Store.Open(OpenFlags flags)
   at Internal.Cryptography.Pal.OpenSslX509ChainProcessor.FindCandidates(X509Certificate2 leaf, X509Certificate2Collection extraStore, HashSet`1 downloaded, HashSet`1 systemTrusted, TimeSpan& remainingDownloadTime)
   at Internal.Cryptography.Pal.ChainPal.BuildChain(Boolean useMachineContext, ICertificatePal cert, X509Certificate2Collection extraStore, OidCollection applicationPolicy, OidCollection certificatePolicy, X509RevocationMode revocationMode, X509RevocationFlag revocationFlag, DateTime verificationTime, TimeSpan timeout)
   at System.Security.Cryptography.X509Certificates.X509Chain.Build(X509Certificate2 certificate, Boolean throwOnException)
   at System.Security.Cryptography.X509Certificates.X509Chain.Build(X509Certificate2 certificate)
   at System.Net.Security.CertificateValidation.BuildChainAndVerifyProperties(X509Chain chain, X509Certificate2 remoteCertificate, Boolean checkCertName, String hostName)
   at System.Net.Security.SecureChannel.VerifyRemoteCertificate(RemoteCertValidationCallback remoteCertValidationCallback, ProtocolToken& alertToken)
   at System.Net.Security.SslState.CompleteHandshake(ProtocolToken& alertToken)
   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReceiveBlob(Byte[] buffer, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)
[... 8 lines cut ...]
--- End of stack trace from previous location where exception was thrown ---
   at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore(Stream stream, SslClientAuthenticationOptions sslOptions, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore(Stream stream, SslClientAuthenticationOptions sslOptions, CancellationToken cancellationToken)
   at System.Threading.Tasks.ValueTask`1.get_Result()
   at System.Net.Http.HttpConnectionPool.CreateConnectionAsync(HttpRequestMessage request, CancellationToken cancellationToken)
   at System.Threading.Tasks.ValueTask`1.get_Result()
   at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync(ValueTask`1 creationTask)
   at System.Threading.Tasks.ValueTask`1.get_Result()
   at System.Net.Http.HttpConnectionPool.SendWithRetryAsync(HttpRequestMessage request, Boolean doRequestAuth, CancellationToken cancellationToken)
   at System.Net.Http.RedirectHandler.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken)
   at System.Net.Http.HttpClient.FinishSendAsyncBuffered(Task`1 sendTask, HttpRequestMessage request, CancellationTokenSource cts, Boolean disposeCts)
[... cut ...] llationToken)
   at OpcPublisher.IotHubCommunication.InitAsync()
```

BTW: We have build the publisher for linux-arm as described in https://github.com/dotnet/core/blob/master/samples/RaspberryPiInstructions.md . And after the build, the generated files had been copied from a Windows file system to a case sensitive Linux file system.
So I faced a problem due to rename of `OpcPublisher.dll` into `opcpublisher.dll`.
Could the issue above also be caused by some thing similar?

Greetings
_   Juergen

hansgschossmann, 2019-02-13:
@JuergenKosel Can you check if the DigiCert Baltimore Root CA Certificate is in the store?
Can you give me some more info on the case issue or submit a PR?

JuergenKosel, 2019-02-14:
Hello @hansgschossmann ,

the problem was an incorrect update of the dotnet runtime environment.
Now I extract the arm binaries from your docker image with a helper docker image:
Dockerfile
```
# Docker image to copy opc publisher from docker image for arm architexture to a volume
# See https://ownyourbits.com/2018/06/27/running-and-building-arm-docker-containers-in-x86/
# docker container run --rm --name publisher-extract -w $(pwd) -v $(pwd):$(pwd) opc-publisher:2.3.2-linux-arm32v7-2amd64

FROM mcr.microsoft.com/iotedge/opc-publisher:2.3.2-linux-arm32v7

COPY qemu-arm-static /usr/bin

ENTRYPOINT ["cp", "-r",  "/app", "/usr/share/dotnet", "./" ]

```
So I can copy the binaries from an image which was build for linux-arm.

closed 2019-02-14 by JuergenKosel