Azure/iot-edge-opc-publisher-v1 #196: RetryPolicy for output messages?

BewaControl-ReneDivossen, 2019-02-14:
What is the default RetryPolicy for messages send by the OpcPublisher used as IotEdge module ?
Can the RetryPolicy be changed (e.g. using commandline parameters) ?

hansgschossmann, 2019-02-15:
it does use in any case (Edge and non Edge):
                ExponentialBackoff exponentialRetryPolicy = new ExponentialBackoff(int.MaxValue, TimeSpan.FromMilliseconds(2), TimeSpan.FromMilliseconds(1024), TimeSpan.FromMilliseconds(3));
makes sense?

BewaControl-ReneDivossen, 2019-02-18:
@hansgschossmann How long will the publisher retry to send messages, i.e. what is the value of **DefaultOperationTimeoutInMilliseconds** ?

hansgschossmann, 2019-02-18:
I have not tried this, but the definition of ExponentialBackoff is:
        public ExponentialBackoff(int retryCount, TimeSpan minBackoff, TimeSpan maxBackoff, TimeSpan deltaBackoff);
It will try int.MaxValue times using the given backoff parameters.

BewaControl-ReneDivossen, 2019-02-18:
@hansgschossmann If **DefaultOperationTimeoutInMilliseconds** is not set within OpcPublisher, the default should be 4 minutes. 

> The retry mechanism stops after DefaultOperationTimeoutInMilliseconds, which is currently set at 4 minutes.

([source](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-reliability-features-in-sdks#net-implementation-guidance))

Reading the linked source, it is unclear to me whether the 4 minute limit is always applicable or only for "throttling errors".

hansgschossmann, 2019-02-18:
i see. i will tag this issue as feature request. adding a command line option to configure it should to the job.

BewaControl-ReneDivossen, 2019-02-18:
OK - thank you.