Azure/iot-edge-opc-publisher-v1 #197: Could not send an Open Secure Channel request.

Bodera, 2019-02-14:
Hello people,

  I discover the project this week and I'm still tring to get know how perform telemetry from a OPC UA Server to Azure IoT Hub, but I'm think that I'm more close than ever. My problem maybe related to the issue #53.

  I clonned the master branch and compile the project in Visual Studio, I don't know yet the reason for don't output a .exe file, just the dll. This is not a problem, after read the docs I understood how run the module by using PowerShell commands.

  My actual situation is the output:
```
[12:06:57 INF] Publisher is running. Press CTRL-C to quit.
[12:06:57 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://127.0.0.1:49310'.
[12:07:00 INF] Create secured session for endpoint URI 'opc.tcp://127.0.0.1:49310' with timeout of 10000 ms.
[12:07:02 ERR] Session creation to endpoint 'opc.tcp://127.0.0.1:49310' failed 1 time(s). Please verify if server is up and Publisher configuration is correct.
Opc.Ua.ServiceResultException: Error establishing a connection: Could not send an Open Secure Channel request.
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 888
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in C:\Users\rafae\Documents\IGS\publisher\iot-edge-opc-publisher\opcpublisher\OpcSession.cs:line 295
```

After reading the error message I've checked my OPC Server status and it's good and running.
So bellow I present my `publishernodes.json` config file which I presume that is the file that error message points to me.
> Publisher configuration is correct.
```json
[
  {
    "EndpointUrl": "opc.tcp://127.0.0.1:49310",
    "OpcNodes": [
      {
        "Id": "s=Simulation Examples.Functions.User2",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 5000,
        "SkipFirst": true
      }
    ]
  }
]
```
I thought that's is enough info for propose a diagnostics, but will be my pleasure provide more in depth details for contribut in find the solution. Thanks you all for maintaining this tool, I like work on it.

hansgschossmann, 2019-02-15:
pls check my comment here: https://github.com/Azure/iot-edge-opc-publisher/issues/198

hansgschossmann, 2019-02-18:
closing for now. pls reopen or open another issue if it is still not working.

closed 2019-02-18 by hansgschossmann