Azure/iot-edge-opc-publisher-v1 #198: messages are not reaching to iothub

lakshmisivareddy, 2019-02-15:
Hi i am tying on windows with opc publisher i am using the below command 

dotnet opcpublisher.dll opcua HostName=<connectionstring> --tc telemetryconfiguration.json --di=5
telemetry.json:
{
  "Defaults": {
    "EndpointUrl": {
      "Publish": false,
      "Pattern": "(.*)",
      "Name": "EndpointUrl"
    },
    "NodeId": {
      "Publish": true,
      "Name": "NodeId"
    },
    "MonitoredItem": {
      "Flat": true,
      "ApplicationUri": {
        "Publish": false,
        "Name": "ApplicationUri"
      },
      "DisplayName": {
        "Publish": false,
        "Name": "DisplayName"
      }
    },
    "Value": {
      "Flat": true,
      "Value": {
        "Publish": true,
        "Name": "Value"
      },
      "SourceTimestamp": {
        "Publish": true,
        "Name": "SourceTimestamp"
      },
      "StatusCode": {
        "Publish": true,
        "Name": "StatusCode"
      },
      "Status": {
        "Publish": false,
        "Name": "Status"
      }
    }
  }
}

publishednodes.json:
[
  {
    "EndpointUrl": "opc.tcp://172.30.8.102:4840",
    "OpcNodes": [
      {
        "Id": "i=20305",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 5000,
        "SkipFirst": false
      }
    ]
  }
]

i am getting the below message in log file 
2019-02-15 20:01:04.676 +05:30 [INF] messages sent to IoTHub: 0
2019-02-15 20:01:04.677 +05:30 [INF] last successful msg sent @: 1/1/0001 12:00:00 AM
2019-02-15 20:01:04.680 +05:30 [INF] bytes sent to IoTHub: 0
2019-02-15 20:01:04.682 +05:30 [INF] avg msg size: 0
2019-02-15 20:01:04.685 +05:30 [INF] msg send failures: 0
2019-02-15 20:01:04.687 +05:30 [INF] messages too large to sent to IoTHub: 0
2019-02-15 20:01:04.688 +05:30 [INF] times we missed send interval: 0
2019-02-15 20:01:04.689 +05:30 [INF] number of events: 0

hansgschossmann, 2019-02-15:
first thing: on windows native or windows containers you need to use --at X509Store
try that and post the log when OPC Publisher is starting up

lakshmisivareddy, 2019-02-18:
Hi @hansgschossmann 
i tried as you mentioned
dotnet opcpublisher.dll opcua  <iothub connection string>--tc tc.json --di=5   --at X509Store
but still getting same problem 

2019-02-18 11:15:04.671 +05:30 [INF] Loaded 1 config file entry/entries.
2019-02-18 11:15:04.677 +05:30 [INF] There are 1 nodes to publish.
2019-02-18 11:15:04.680 +05:30 [INF] initialize hub communication.
2019-02-18 11:15:04.691 +05:30 [INF] IoTHub device cert store type is: X509Store
2019-02-18 11:15:04.694 +05:30 [INF] IoTHub device cert path is: My
2019-02-18 11:15:04.697 +05:30 [INF] Attempting to register ourselves with IoT Hub using owner connection string.
2019-02-18 11:15:06.553 +05:30 [INF] Device 'opcua' found in IoTHub registry. Remove it.
2019-02-18 11:15:06.848 +05:30 [INF] Adding device 'opcua' to IoTHub registry.
2019-02-18 11:15:07.382 +05:30 [INF] Generate device connection string.
2019-02-18 11:15:07.388 +05:30 [INF] Adding device connectionstring to secure store.
2019-02-18 11:15:08.632 +05:30 [INF] Attempting to read device connection string from cert store using subject name: opcua
2019-02-18 11:15:08.650 +05:30 [INF] Protocol used for IoTHub communication is'Mqtt_WebSocket_Only'
2019-02-18 11:15:08.695 +05:30 [INF] IoTCentral mode: False
2019-02-18 11:15:12.220 +05:30 [INF] Connection status changed to 'Connected', reason 'Connection_Ok'
2019-02-18 11:15:12.833 +05:30 [INF] Message processing and hub communication configured with a send interval of 10 sec and a message buffer size of 262144 bytes.
2019-02-18 11:15:12.839 +05:30 [INF] Creating task process and batch monitored item data updates...
2019-02-18 11:15:12.903 +05:30 [INF] initialize publisher diagnostics
2019-02-18 11:15:12.907 +05:30 [INF] DiagnosticsInterval :: 5
2019-02-18 11:15:12.909 +05:30 [INF] 
2019-02-18 11:15:12.914 +05:30 [INF] 
2019-02-18 11:15:12.917 +05:30 [INF] Publisher is running. Press CTRL-C to quit.
2019-02-18 11:15:12.926 +05:30 [INF] ShowDiagnosticsInfoAsync
2019-02-18 11:15:12.944 +05:30 [INF] Connect and monitor session and nodes on endpoint 'opc.tcp://172.30.8.102:4840'.
2019-02-18 11:15:15.075 +05:30 [INF] Create secured session for endpoint URI 'opc.tcp://172.30.8.102:4840' with timeout of 10000 ms.
2019-02-18 11:15:15.426 +05:30 [INF] Session successfully created with Id ns=1;g=a2a48d90-3646-645f-7c47-74f707f09f39.
2019-02-18 11:15:15.448 +05:30 [INF] The session to endpoint 'opc.tcp://172.30.8.102:4840/' has 2 entries in its namespace array:
2019-02-18 11:15:15.449 +05:30 [INF] Namespace index 0: http://opcfoundation.org/UA/
2019-02-18 11:15:15.450 +05:30 [INF] Namespace index 1: urn:unconfigured:application
2019-02-18 11:15:15.533 +05:30 [INF] Created subscription with id 1 on endpoint 'opc.tcp://172.30.8.102:4840'
2019-02-18 11:15:15.539 +05:30 [INF] Create subscription on endpoint 'opc.tcp://172.30.8.102:4840' requested OPC publishing interval is 5000 ms. (revised: 5000 ms)
2019-02-18 11:15:15.547 +05:30 [INF] Start monitoring items on endpoint 'opc.tcp://172.30.8.102:4840'. Currently monitoring 0 items.
2019-02-18 11:15:15.672 +05:30 [INF] Done processing unmonitored items on endpoint 'opc.tcp://172.30.8.102:4840' took 119 msec. Now monitoring 1 items in subscription with id '1'.
2019-02-18 11:15:18.013 +05:30 [INF] ==========================================================================
2019-02-18 11:15:18.017 +05:30 [INF] OpcPublisher status @ 2/18/2019 5:45:18 AM (started @ 2/18/2019 5:45:02 AM)
2019-02-18 11:15:18.021 +05:30 [INF] ---------------------------------
2019-02-18 11:15:18.026 +05:30 [INF] OPC sessions (configured/connected): 1/1
2019-02-18 11:15:18.032 +05:30 [INF] OPC subscriptions (configured/connected): 1/1
2019-02-18 11:15:18.039 +05:30 [INF] OPC monitored items (configured/monitored/to remove): 1/1/0
2019-02-18 11:15:18.043 +05:30 [INF] ---------------------------------
2019-02-18 11:15:18.048 +05:30 [INF] monitored items queue bounded capacity: 8192
2019-02-18 11:15:18.055 +05:30 [INF] monitored items queue current items: 0
2019-02-18 11:15:18.059 +05:30 [INF] monitored item notifications enqueued: 0
2019-02-18 11:15:18.064 +05:30 [INF] monitored item notifications enqueue failure: 0
2019-02-18 11:15:18.070 +05:30 [INF] ---------------------------------
2019-02-18 11:15:18.075 +05:30 [INF] messages sent to IoTHub: 0
2019-02-18 11:15:18.081 +05:30 [INF] last successful msg sent @: 1/1/0001 12:00:00 AM
2019-02-18 11:15:18.090 +05:30 [INF] bytes sent to IoTHub: 0
2019-02-18 11:15:18.095 +05:30 [INF] avg msg size: 0
2019-02-18 11:15:18.098 +05:30 [INF] msg send failures: 0
2019-02-18 11:15:18.102 +05:30 [INF] messages too large to sent to IoTHub: 0
2019-02-18 11:15:18.109 +05:30 [INF] times we missed send interval: 0
2019-02-18 11:15:18.113 +05:30 [INF] number of events: 0
2019-02-18 11:15:18.116 +05:30 [INF] ---------------------------------
2019-02-18 11:15:18.122 +05:30 [INF] current working set in MB: 91
2019-02-18 11:15:18.127 +05:30 [INF] --si setting: 10
2019-02-18 11:15:18.132 +05:30 [INF] --ms setting: 262144
2019-02-18 11:15:18.138 +05:30 [INF] --ih setting: Mqtt_WebSocket_Only
2019-02-18 11:15:18.141 +05:30 [INF] ==========================================================================
2019-02-18 11:15:23.160 +05:30 [INF] ==========================================================================
2019-02-18 11:15:23.164 +05:30 [INF] OpcPublisher status @ 2/18/2019 5:45:23 AM (started @ 2/18/2019 5:45:02 AM)
2019-02-18 11:15:23.168 +05:30 [INF] ---------------------------------
[... cut ...]
2019-02-18 11:15:23.216 +05:30 [INF] ---------------------------------
2019-02-18 11:15:23.224 +05:30 [INF] current working set in MB: 91
2019-02-18 11:15:23.229 +05:30 [INF] --si setting: 10
2019-02-18 11:15:23.234 +05:30 [INF] --ms setting: 262144
2019-02-18 11:15:23.241 +05:30 [INF] --ih setting: Mqtt_WebSocket_Only
2019-02-18 11:15:23.245 +05:30 [INF] ==========================================================================
2019-02-18 11:15:28.262 +05:30 [INF] ==========================================================================

hansgschossmann, 2019-02-18:
you configure the following node to be published:
"Id": "i=20305",
if that is in your namespace with index 1 "urn:unconfigured:application" then you need to configure it as:
"Id": "ns=1;i=20305",

best way to find this out is to connect to your OPC UA server with an client and check the details for the node. it specifies the namespace index in which your node lives.

lakshmisivareddy, 2019-02-18:
Hi @hansgschossmann 
thanks for the support  it is sending messages now 

2019-02-18 20:19:25.491 +05:30 [INF] ---------------------------------
2019-02-18 20:19:25.493 +05:30 [INF] messages sent to IoTHub: 1
2019-02-18 20:19:25.495 +05:30 [INF] last successful msg sent @: 2/18/2019 2:49:20 PM
2019-02-18 20:19:25.497 +05:30 [INF] bytes sent to IoTHub: 102
2019-02-18 20:19:25.499 +05:30 [INF] avg msg size: 102
2019-02-18 20:19:25.500 +05:30 [INF] msg send failures: 0
2019-02-18 20:19:25.503 +05:30 [INF] messages too large to sent to IoTHub: 0
2019-02-18 20:19:25.505 +05:30 [INF] times we missed send interval: 0
2019-02-18 20:19:25.507 +05:30 [INF] number of events: 2
2019-02-18 20:19:25.509 +05:30 [INF] ---------------------------------
2019-02-18 20:19:25.511 +05:30 [INF] current working set in MB: 90
2019-02-18 20:19:25.513 +05:30 [INF] --si setting: 10
2019-02-18 20:19:25.515 +05:30 [INF] --ms setting: 262144
2019-02-18 20:19:25.517 +05:30 [INF] --ih setting: Mqtt_WebSocket_Only
2019-02-18 20:19:25.519 +05:30 [INF] ==========================================================================
2019-02-18 20:19:30.164 +05:30 [INF] encodedhubMessage305
2019-02-18 20:19:30.165 +05:30 [INF] hubMessageMicrosoft.Azure.Devices.Client.Message
2019-02-18 20:19:30.528 +05:30 [INF] ==========================================================================
2019-02-18 20:19:30.529 +05:30 [INF] OpcPublisher status @ 2/18/2019 2:49:30 PM (started @ 2/18/2019 2:48:58 PM)
2019-02-18 20:19:30.531 +05:30 [INF] ---------------------------------
2019-02-18 20:19:30.532 +05:30 [INF] OPC sessions (configured/connected): 1/1
2019-02-18 20:19:30.533 +05:30 [INF] OPC subscriptions (configured/connected): 1/1
2019-02-18 20:19:30.534 +05:30 [INF] OPC monitored items (configured/monitored/to remove): 1/1/0
2019-02-18 20:19:30.535 +05:30 [INF] ---------------------------------
2019-02-18 20:19:30.536 +05:30 [INF] monitored items queue bounded capacity: 8192
2019-02-18 20:19:30.538 +05:30 [INF] monitored items queue current items: 0
2019-02-18 20:19:30.539 +05:30 [INF] monitored item notifications enqueued: 3
2019-02-18 20:19:30.541 +05:30 [INF] monitored item notifications enqueue failure: 0
2019-02-18 20:19:30.543 +05:30 [INF] ---------------------------------
2019-02-18 20:19:30.545 +05:30 [INF] messages sent to IoTHub: 2
2019-02-18 20:19:30.547 +05:30 [INF] last successful msg sent @: 2/18/2019 2:49:30 PM
2019-02-18 20:19:30.548 +05:30 [INF] bytes sent to IoTHub: 305
2019-02-18 20:19:30.552 +05:30 [INF] avg msg size: 152
2019-02-18 20:19:30.556 +05:30 [INF] msg send failures: 0
2019-02-18 20:19:30.558 +05:30 [INF] messages too large to sent to IoTHub: 0
2019-02-18 20:19:30.560 +05:30 [INF] times we missed send interval: 0
2019-02-18 20:19:30.562 +05:30 [INF] number of events: 3
2019-02-18 20:19:30.564 +05:30 [INF] ---------------------------

hansgschossmann, 2019-02-18:
great to hear.

closed 2019-02-18 by hansgschossmann