Azure/iot-edge-opc-publisher-v1 #203: communication with IoT hub stops after some hours.

anhame, 2019-02-20:
Hello,
I'm using the OpcPublisher with my IoT hub as a standalone application and the Mqtt protocol.
My problem is that after a few hours of operation I'm receiving the following error:
**Connection status changed to 'Disconnected', reason 'Retry_Expired'**
After this, the communication with IoT hub is completely lost and the msg send failures counter in the OpcPublisher status output starts to increase.
Before there were also several messages (repeated almost hourly) like this:
**Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error'**

What could be reason for this? There were no such problems occurring in former versions of the publisher (half a year ago).

Any help appreciated.
Thanks
Hartmut

hansgschossmann, 2019-02-21:
are you using the code of the latest tag 2.3.3? this issue is bubbling up from IoT SDK. First thing i would try if you use the latest code is try to switch to Amqp_Websocket_Only

anhame, 2019-02-22:
Yes, I'm using the latest version 2.3.3. I switched to Amqp_Websocket_Only and it seems to be better (still receiving [ERR] Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error', but not that frequently as before (one or two per day). So I guess I will finally run into the same problem, but only later... ?

hansgschossmann, 2019-02-22:
Let it run and update the issue if you hit the problem again.

anhame, 2019-03-11:
We discovered the same issue with Amqp_WebSocket_Only as well. It just doesn't happens as frequently as when using MQTT.
2019-03-11 01:28:46.512 +01:00 [INF] ==========================================================================
2019-03-11 01:30:56.320 +01:00 [ERR] Connection status changed to 'Disconnected', reason 'Retry_Expired'
2019-03-11 01:33:46.510 +01:00 [INF] ==========================================================================

Anyhow, is it really necessary to stop the communication after a certain number of unsuccessful retries?

anhame, 2019-07-02:
With Devices.Client version 1.20.1 there is a slightly different behavior, it first reports the following for several times:
2019-06-13 23:24:22.941 +02:00 [ERR] Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error'
2019-06-13 23:24:51.081 +02:00 [INF] Connection status changed to 'Connected', reason 'Connection_Ok'

After about 60 disconnected/retries which take place over time, there is one final status changed message:

2019-06-13 23:58:59.972 +02:00 [ERR] Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error'
2019-06-13 23:59:19.722 +02:00 [ERR] Connection status changed to 'Disconnected', reason 'Communication_Error'

After this, the communication won't be established again and I do have to restart the opc publisher.
What can be causing this and how could it be solved?

Any help appreciated!