Azure/iot-edge-opc-publisher-v1 #204: Cannot connect to OPCUA Server with hostname in server endpoint

CHBrianLiu, 2019-02-25:
Hi, 

I confronted a problem while I was using **publisher ver. 2.3.3** as an IoT Edge module on Azure VM (Windows 10 Pro 1809). When I specified the server endpoint with **hostname** in **pn.json** file, the publisher module couldn't establish connection with OPCUA server (I have done additional test that I replaced **hostname** with **IP address**, and it worked!). Could you please advise me how I can fix the connection problem with **hostname** in server endpoint? 

And there was an **ExtraHosts** container create option to specify the hostname-IP pairs in the previous version. May I know how publisher module looks up the IP without **ExtraHosts** configuration?
And I provide my system information and module settings below. 

While using **hostname** in server endpoint (*opc.tcp://brianopc:62541/Quickstarts/ReferenceServer*), the prompted error messages are as following.
```
[07:41:12 INF] Publisher is running. Press CTRL-C to quit.
[07:41:12 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://brianopc:62541/Quickstarts/ReferenceServer'.
[07:41:36 ERR] Session creation to endpoint 'opc.tcp://brianopc:62541/Quickstarts/ReferenceServer' failed 1 time(s). Please verify if server is up and Publisher configuration is correct.
Opc.Ua.ServiceResultException: Error establishing a connection: BadNotConnected
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request)
   at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints)
   at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris)
   at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(String discoveryUrl, Boolean useSecurity, Int32 operationTimeout)
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in C:\app\opcpublisher\OpcSession.cs:line 288
```
However, I changed the **hostname (brianopc)** in the server endpoint to my **VM's IP address (10.0.3.4)**, the publisher module worked fine! And the messages are stated as below.
```
[07:48:25 INF] Publisher is running. Press CTRL-C to quit.
[07:48:25 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer'.
[07:48:27 INF] Create unsecured session for endpoint URI 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer' with timeout of 10000 ms.
[07:48:27 INF] Session successfully created with Id ns=3;i=2093901607.
[07:48:27 INF] The session to endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer' has 4 entries in its namespace array:
[07:48:27 INF] Namespace index 0: http://opcfoundation.org/UA/
[07:48:27 INF] Namespace index 1: urn:brianopc:UA:Quickstarts:ReferenceServer
[07:48:27 INF] Namespace index 2: http://opcfoundation.org/Quickstarts/ReferenceApplications
[07:48:27 INF] Namespace index 3: http://opcfoundation.org/UA/Diagnostics
[07:48:27 INF] Created subscription with id 3 on endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer'
[07:48:27 INF] Create subscription on endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer' requested OPC publishing interval is 5000 ms. (revised: 5000 ms)
[07:48:27 INF] Start monitoring items on endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer'. Currently monitoring 0 items.
[07:48:28 INF] Done processing unmonitored items on endpoint 'opc.tcp://10.0.3.4:62541/Quickstarts/ReferenceServer' took 916 msec. Now monitoring 1 items in subscription with id '3'.
```
Here I provide my environment info. 
**VM on Azure**
* VM Size: `D2s v3`
* OS: `Windows 10 Pro ver.1809`
* Container: `Windows container`

**IoT Edge & Module** 
* Container URI: `mcr.microsoft.com/iotedge/opc-publisher:2.3.3-windows-amd64`
* Container Create Option:
```json
{
  "Hostname": "publisher",
  "Cmd": [
    "--pf=./pn.json",
    "--aa"
  ],
  "HostConfig": {
    "Binds": [
      "c:/iiotedge:c:/appdata"
    ]
  }
}
```
* pn.json file content:
```json
[
  {
    "EndpointUrl": "opc.tcp://brianopc:62541/Quickstarts/ReferenceServer",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=2;s=Scalar_Simulation_Int64",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 5000,
        "DisplayName": "Current time"
      }
    ]
  }
]
```
Best regards, 
Brian Liu

hansgschossmann, 2019-02-25:
@CHBrianLiu i was simplifying the sample CreateContainerOptions. You can still use the ExtraHosts configuration. I have updated the readme to show how to use it.

markusstuhler, 2019-04-30:
Hi @CHBrianLiu, can you confirm that the updated documentation fixes the issue?