Azure/iot-edge-opc-publisher-v1 #205: Bug in the PublisherTelemetryConfiguration.cs file - Deserializing Interface Properties

marru8619, 2019-02-26:
Hi Everybody,

I Discover a problem with the new issue of iot-edge-opc-publisher. The problem is related to the Deserializing interface properties in the "PublisherTelemetryConfiguration.cs" row 600:

 _telemetryConfiguration = JsonConvert.DeserializeObject<ITelemetryConfigurationFileModel>(await File.ReadAllTextAsync(PublisherTelemetryConfigurationFilename).ConfigureAwait(false));

Every time that I configure the telemetry file or use the default telemetry in the release, the program goes fail and the message is:

**Loading of the telemetry configuration file failed. Does the file exist and has correct syntax? Exiting...
Newtonsoft.Json.JsonSerializationException: Could not create an instance of type OpcPublisher.ITelemetryConfigurationFileModel. Type is an interface or abstract class and cannot be instantiated. Path 'Defaults', line 23, position 15.**

Of course the file exist and is configured correctly as per default.

Take note that with the previous version of OPC Publisher solution it work fine.

Some one know hot to fix or how to contact the support?

Eichi87, 2019-03-05:
exactly the same problem here. no solution right now. :-/

marru8619, 2019-03-05:
Hi Eichi87, 

thank you now I know that I'm not alone. 

Do you have also problem regard the communication? I mean that after 1 hour (more or less) that the publisher is active, appear a message that say:

**2019-02-11 09:49:25.572 +01:00 [ERR] Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error'
2019-02-11 09:49:25.802 +01:00 [INF] Connection status changed to 'Connected', reason 'Connection_Ok'**

and seems like that the publisher is freezed?

Eichi87, 2019-03-07:
I will test that

marru8619, 2019-03-08:
Hi Eichi87,

I'm using the current version and that happen on it. 

This is my personal mail in case we wont to have a private conversation about this : mrusso8619@gmail.com

Thank you

lucianoctorres, 2019-03-18:
Any information on this? Just updated the images in our IoT clients and now it's broken...

marru8619, 2019-03-18:
Hi lucianoctorres,

nothing new. I think there is a bug in the software (the interface is wrong).

marru8619, 2019-03-21:
I close the issue, they fix the problem. Now it work

closed 2019-03-21 by marru8619