Azure/iot-edge-opc-publisher-v1 #212: ERROR: Could not create module publisher: invalid reference format

ddobric, 2019-03-14:
Hi all,

trying to deploy _opcua_ publisher module to IOTEDGE and getting following error:

```
System.AggregateException: One or more errors occurred. (Error calling Create module publisher: Could not create module publisher
        caused by: Could not pull image docker pull mcr.microsoft.com/iotedge/opc-publisher:2.3
        caused by: invalid reference format) ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException: Error calling Create module publisher: Could not create module publisher
        caused by: Could not pull image docker pull mcr.microsoft.com/iotedge/opc-publisher:2.3
        caused by: invalid reference format
```

Running Microsoft Windows 10 Enterprise LTSC with (WINDOWS) container support.

ddobric, 2019-03-17:
?

SachaLhopital, 2019-06-19:
It seems to be a `docker pull` error. 
Can you try it with the `latest` docker image instead of the 2.3 ? 
Can you try to pull the image yourself ?