Azure/iot-edge-opc-publisher-v1 #213: Why is device recreated by publisher?

ddobric, 2019-03-18:
Hi all,

would like to know why, publisher delete and then create device on startup?
When running multiple publisher in cluster (fail-over or LB), this is an issue. Moreover, publisher requires for this operation higher permission set than device itself.

Thanks in advance.

anhame, 2019-03-19:
You can specify a connection string via command line. This way, the publisher won't recreate the device.

> --dc,  --deviceconnectionstring=VALUE
>                                 if publisher is not able to register itself with
>                                   IoTHub, you can create a device with name <
>                                   applicationname> manually and pass in the
>                                   connectionstring of this device.
>                                   Default: none

ddobric, 2019-04-09:
Thanks. This works.
It would be great to update documentation, so we can peek the right configuration easier.

closed 2019-04-09 by ddobric