Azure/iot-edge-opc-publisher-v1 #214: Unable to query telemetry

bialad, 2019-03-20:
I've tried to process the published telemetry through IoT Hub message routing, but I can't query properties or body using these queries.

`$iothub-connection-module-id = 'OpcPublisher'`
`IS_DEFINED($body.NodeId)`

[According to this it should work](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-devguide-routing-query-syntax)

```
"OpcPublisher": {
    "version": "1.0",
    "type": "docker",
    "status": "running",
    "restartPolicy": "always",
    "settings": {
        "image": "mcr.microsoft.com/iotedge/opc-publisher:latest",
        "createOptions": {
            "Hostname": "publisher",
            "Cmd": [
                "publisher",
                "--aa"
            ],
            "HostConfig": {
                "PortBindings": {
                    "62222/tcp": [
                        {
                            "HostPort": "62222"
                        }
                    ]
                },
                "Binds": [
                    "x509certstores:/root/.dotnet/corefx/cryptography/x509stores",
                    "/home/csi/opcpublisher:/appdata"
                ]
            }
        }
    }
}
```

Is there something special with the message format of the opc-publisher? Here's a sample?

```
{
  "message": {
    "annotations": {
      "iothub-connection-device-id": "iot-edge-norden",
      "iothub-connection-module-id": "OpcPublisher",
      "iothub-connection-auth-method": "{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}",
      "iothub-connection-auth-generation-id": "636885143300408265",
      "iothub-enqueuedtime": 1553078774722,
      "iothub-message-source": "Telemetry",
      "x-opt-sequence-number": 359,
      "x-opt-offset": "345520",
      "x-opt-enqueued-time": 1553078774925
    },
    "applicationProperties": undefined,
    "body": [
      {
        "NodeId": "ns=2;s=NordenApplication.GVL_Fun06_11_HASeal.G_iTempWaterOut",
        "ApplicationUri": "192.168.165.11/SchneiderElectric/Uaserver",
        "DisplayName": "ns=2;s=NordenApplication.GVL_Fun06_11_HASeal.G_iTempWaterOut",
        "Value": {
          "Value": 28,
          "SourceTimestamp": "2019-03-20T11:32:29.8940000Z"
        }
      },
      {
        "NodeId": "ns=2;s=NordenApplication.GVL_Sys01_Schneider.G_iTasksAliveCounter",
        "ApplicationUri": "192.168.165.11/SchneiderElectric/Uaserver",
        "DisplayName": "ns=2;s=NordenApplication.GVL_Sys01_Schneider.G_iTasksAliveCounter",
        "Value": {
          "Value": 272,
          "SourceTimestamp": "2019-03-20T11:32:29.8940000Z"
        }
      }
    ]
  }
}
```

bialad, 2019-03-20:
OK, so i figured out that `iothub-connection-module-id` for some reason gets changed into `connectionModuleId`, so `$connectionModuleId = "OpcPublisher"` works.

Question still stands how to filter on the message body?

markusstuhler, 2019-05-02:
Hi @bialad, glad to hear that you could solve the first issue like you described. About the query in the message body: messages are batched, therefore the message body is a json array, that cannot be queried directly. However, could you try if it works when you disable batching by running the publisher with the command line option "--ms=0 --si=0"? That forces the publisher to send each message directly. Thanks!

bialad, 2019-05-20:
I've moved on to other projects, and am unable to do further testing.

closed 2019-05-20 by bialad