Azure/iot-edge-opc-publisher-v1 #221: BadSubscriptionIdInvalid for nodes with SamplingInterval of 100ms

dcrupel, 2019-04-01:
Hi,

[image]

After running without problems for months the publisher stops monitoring the Tags with a sampling interval of 100ms, the ones with sampling interval of 1000ms keep coming in.

So, the subscription of the 100ms nodes is stopped, the other one (1000ms sampling interval) remains.

Any idea how to prevent this from happening? 

Thanks!

Regards

markusstuhler, 2019-04-30:
Hi @dcrupel,  we'll look into this.

markusstuhler, 2019-05-02:
Hi @dcrupel, could you share which version of the publisher you were using? Did you update at some point in time? Are there any other things that changed?

krenting, 2019-06-20:
Hi @dcrupel,

I had this same problem but only when i connected to the machine via middleware (Kepware). When connected to the machine directly i could use sample rates of 50ms without any problems. Have you done some testing with different OPC UA servers?

Regards

dcrupel, 2019-06-20:
Yes, It is a Kepware problem so this has nothing to do with the OPC publisher. I'll close this issue.

closed 2019-06-20 by dcrupel

krenting, 2019-06-20:
I'm in contact with the developers of Kepware on this issue. If there is any progress I'll send you an update.