Azure/iot-edge-opc-publisher-v1 #223: Not all nodes sending data to IoTHub

DunRemko, 2019-04-05:
We have a running OPCPublisher with the following pn.json file:
[
  {
    "EndpointUrl": "opc.tcp:// x.x.x.x:xxxx ",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=2;s=DL_01.NI_0001.HValue"
      },
      {
        "Id": "ns=2;s=DL_01.NI_0001.Value"
      },
      {
        "Id": "ns=2;s=DL_01.NI_0002.HValue"
      },
      {
        "Id": "ns=2;s=DL_01.NI_0002.Value"
      },
      {
        "Id": "ns=2;s=IP.Failed"
      },
      {
        "Id": "ns=2;s=ControlKey.NumberOfDataLogger"
      }
    ]
  }
]

In publisher-publisher.log we see that 6 items are montitored.
But only 2 of these are sending data to the Iot Hub:
ns=2;s=IP.Failed
ns=2;s=ControlKey.NumberOfDataLogger

2019-04-05 08:08:05.045 +00:00 [DBG] OPC: SECURE CHANNEL CREATED [.NetStandard ClientChannel UA-TCP 1.04.354.23] [ID=6] Connected To: opc.tcp:// x.x.x.x:xxxx / [None/None/Binary]
2019-04-05 08:08:05.076 +00:00 [INF] Session successfully created with Id ns=1;g=5f234975-0a03-463f-a612-0837bb15864d.
2019-04-05 08:08:05.079 +00:00 [INF] The session to endpoint 'opc.tcp://x.x.x.x:xxxx/' has 3 entries in its namespace array:
2019-04-05 08:08:05.080 +00:00 [INF] Namespace index 0: http://opcfoundation.org/UA/
2019-04-05 08:08:05.080 +00:00 [INF] Namespace index 1: urn:xxxxxxx:xxxx
2019-04-05 08:08:05.080 +00:00 [INF] Namespace index 2: http:// xxxxxxx / xxxxxxx / xxxxxxx /
2019-04-05 08:08:05.114 +00:00 [INF] Created subscription with id 1915832 on endpoint 'opc.tcp:// x.x.x.x:xxxx '
2019-04-05 08:08:05.114 +00:00 [INF] Create subscription on endpoint 'opc.tcp:// x.x.x.x:xxxx ' requested OPC publishing interval is 0 ms. (revised: 0 ms)
2019-04-05 08:08:05.115 +00:00 [INF] Start monitoring items on endpoint 'opc.tcp:// x.x.x.x:xxxx. Currently monitoring 0 items.
2019-04-05 08:08:05.137 +00:00 [INF] Done processing unmonitored items on endpoint 'opc.tcp:// x.x.x.x:xxxx ' took 22 msec. Now monitoring 6 items in subscription with id '1915832'.
2019-04-05 08:08:05.248 +00:00 [DBG]    ApplicationUri: 
2019-04-05 08:08:05.248 +00:00 [DBG]    EndpointUrl: opc.tcp:// x.x.x.x:xxxx
2019-04-05 08:08:05.248 +00:00 [DBG]    DisplayName: ns=2;s=IP.Failed
2019-04-05 08:08:05.248 +00:00 [DBG]    Value: false
2019-04-05 08:08:05.248 +00:00 [DBG] Enqueue a new message from subscription 1915832
2019-04-05 08:08:05.248 +00:00 [DBG]  with publishing interval: 0 and sampling interval: 0):
2019-04-05 08:08:05.249 +00:00 [DBG]    ApplicationUri: 
2019-04-05 08:08:05.249 +00:00 [DBG]    EndpointUrl: opc.tcp:// x.x.x.x:xxxx
2019-04-05 08:08:05.249 +00:00 [DBG]    DisplayName: ns=2;s=ControlKey.NumberOfDataLogger
2019-04-05 08:08:05.249 +00:00 [DBG]    Value: 1
2019-04-05 08:08:05.249 +00:00 [DBG] Enqueue a new message from subscription 1915832
2019-04-05 08:08:05.249 +00:00 [DBG]  with publishing interval: 0 and sampling interval: 0):
2019-04-05 08:08:05.280 +00:00 [DBG] Added new message with size 135 to hub message (size is now 136).
2019-04-05 08:08:05.280 +00:00 [DBG] Added new message with size 171 to hub message (size is now 308).

NodeId for 1 of the monitored items which are not sending data in Prosys OPC UA client:
[image]

DunRemko, 2019-04-06:
turned on --fn and --fd : now seeing an error:
2019-04-06 09:26:39.351 +00:00 [DBG] OPC: ***EXCEPTION*** BadNodeIdUnknown
2019-04-06 09:26:39.352 +00:00 [ERR] Failed to monitor node 'ns=1;s=DL_01.NI_0002.HValue' on endpoint 'opc.tcp://xx.x.x.x:xxxx'.
2019-04-06 09:26:39.352 +00:00 [ERR] OPC UA ServiceResultException is 'BadNodeIdUnknown'. Please check your publisher configuration for this node.
2019-04-06 09:26:39.353 +00:00 [DBG] OPC: ***EXCEPTION*** BadNodeIdUnknown

markusstuhler, 2019-05-02:
Hi @DunRemko, could you share the metadata of a node that works so that we can find the "difference" between those two nodes? From logging perspective, everything looks normal. How fequently do the values of the node that don't work change?

DunRemko, 2019-05-14:
After setting Security = true in pn.json and add --to in container create options we are receiving data.

closed 2019-05-14 by DunRemko