Azure/iot-edge-opc-publisher-v1 #226: Publishing subitems of OPC UA Node

HenryVanBragt, 2019-04-10:
I want to communicate subitems of my OPC UA objects to Azure. 
The configuration:
An OPC UA server on a Ewon Flexy contains a node ("s=PulseBit"). This node has several subitems like "isKPI" and "tagID".
Is it possible to communicate this items as subitems of "PulseBit"?

markusstuhler, 2019-05-02:
Hi @HenryVanBragt, automatic publishing of subnodes is currently not on our roadmap.

However, could you share some details on how your setup looks like? How did you model the subnodes on OPC UA side? Are they implemented as OPC UA properties?

HenryVanBragt, 2019-05-02:
Thanks for your reaction. 
I have some more knowledge about OPC UA in the meantime.

I want to sketch my case:
I have a PLC (Siemens S7-1500) with data in a DB 

STRUCT motor1
       INT actualSpeed
       INT setPoint

I have enabled the OPC UA server, so at the OPC UA interface all tags are nodes (motor1, actualSpeed and setPoint). 
When I want to publish Motor1 with it's data to the cloud, I have to configure each node seperated. It would be nice when you can configure to publish Motor1 and all it's referenced nodes, which makes it easier to configure for the PLC programmer