Azure/iot-edge-opc-publisher-v1 #229: Update to latest IotHubSDK

BewaControl-ReneDivossen, 2019-04-17:
Can you please update to the latest IotHubSDK.
There have been some fixes that should make AMQP more stable and should allow reconnecting after failure.

Versions:
Microsoft.Azure.Devices 1.17.3
Microsoft.Azure.Devices.Client 1.20.0

BewaControl-ReneDivossen, 2019-04-23:
@hansgschossmann - Any timeline on this ?

hansgschossmann, 2019-04-23:
@markusstuhler - do you already know when we are going to update?

markusstuhler, 2019-04-30:
This will be part of the 2.3.7 release which we aim to release by the end of this month.

BewaControl-ReneDivossen, 2019-04-30:
@markusstuhler - does "end of this month" mean today ?  Or are you in some part of the world where it is already Mai 1st ?

markusstuhler, 2019-04-30:
My mind was already in the next sprint ;) So I meant end of May.

BewaControl-ReneDivossen, 2019-04-30:
OK. Thanks for clarification.  
I'll be waiting for the release.

BewaControl-ReneDivossen, 2019-05-07:
@markusstuhler - I've seen that you updated the IotHub SDK in the repository.
I'm sorry to be bothering you again but version 1.20.0 is already deprecated again.
There now is 
<PackageReference Include="Microsoft.Azure.Devices.Client" Version="1.20.1" /> 
That's the version they have used in the 1.0.7 release of IotEdge that was released a couple of days ago.

BewaControl-ReneDivossen, 2019-06-13:
Version 1.20.2 is out now and should be referenced.

marcschier, 2019-07-02:
2.4.x is using the latest 1.21.

closed 2019-07-02 by marcschier