Azure/iot-edge-opc-publisher-v1 #230: 'BadEncodingLimitsExceeded ' received when creating subscriptions

jmbrunskill, 2019-04-30:
We have OPC Publisher 2.3.3 running on iotedge 1.0.6.1 (3fa6cbef8b7fc3c55a49a622735eb1021b8a5963)

On startup up of the Publisher we are getting a  'BadEncodingLimitsExceeded ' error.

How should we go about debugging/resolving this issue?

```
[04:34:04 INF] Publisher is running. Press CTRL-C to quit.
[04:34:04 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://10.70.64.95:49320'.
[04:34:06 INF] Create unsecured session for endpoint URI 'opc.tcp://10.70.64.95:49320' with timeout of 10000 ms.
[04:34:06 INF] Session successfully created with Id ns=1;g=73c5b05f-456f-49ec-a62b-a555657aaa6d.
[04:34:06 INF] The session to endpoint 'opc.tcp://10.70.64.95:49320/' has 3 entries in its namespace array:
[04:34:06 INF] Namespace index 0: http://opcfoundation.org/UA/
[04:34:06 INF] Namespace index 1: urn:NZS070HSTOPC01X.prod01.inroot.net:PTC.ThingWorxIC.1:UA%20Server
[04:34:06 INF] Namespace index 2: ThingWorx Industrial Connectivity
[04:34:06 INF] Created subscription with id 8 on endpoint 'opc.tcp://10.70.64.95:49320'
[04:34:06 INF] Create subscription on endpoint 'opc.tcp://10.70.64.95:49320' requested OPC publishing interval is 30000 ms. (revised: 30000 ms)
[04:34:06 INF] Start monitoring items on endpoint 'opc.tcp://10.70.64.95:49320'. Currently monitoring 0 items.
[04:34:06 INF] Now monitoring 10000 items in subscription with id '8'
[04:34:07 INF] Now monitoring 20000 items in subscription with id '8'
[04:34:08 INF] Now monitoring 30000 items in subscription with id '8'
[04:34:10 INF] Now monitoring 40000 items in subscription with id '8'
[04:34:14 INF] Now monitoring 50000 items in subscription with id '8'
[04:34:14 ERR] Unhandled OPC UA ServiceResultException 'BadEncodingLimitsExceeded 'Could not send request to server.'' when monitoring node 'ns=2;s=NZ070_Darfield_NZS070SRV01X.C270.NZ070_B05VA0506_AlmFault' on endpoint 'opc.tcp://10.70.64.95:49320'. Continue.
[04:34:14 INF] Now monitoring 60000 items in subscription with id '8'
[04:34:14 ERR] Unhandled OPC UA ServiceResultException 'BadEncodingLimitsExceeded 'Could not send request to server.'' when monitoring node 'ns=2;s=NZ070_Darfield_NZS070SRV01X.C801.NZ070_S1PW1P01_Msg1' on endpoint 'opc.tcp://10.70.64.95:49320'. Continue.
[04:34:15 ERR] Exception
Opc.Ua.ServiceResultException: Could not send request to server.
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request)
   at Opc.Ua.SessionClient.CreateMonitoredItems(RequestHeader requestHeader, UInt32 subscriptionId, TimestampsToReturn timestampsToReturn, MonitoredItemCreateRequestCollection itemsToCreate, MonitoredItemCreateResultCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Subscription.CreateItems()
   at Opc.Ua.Client.Subscription.ApplyChanges()
   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in /app/opcpublisher/OpcSession.cs:line 561
[04:34:25 INF] Start monitoring items on endpoint 'opc.tcp://10.70.64.95:49320'. Currently monitoring 66541 items.
[04:34:25 ERR] Unhandled OPC UA ServiceResultException 'BadEncodingLimitsExceeded 'Could not send request to server.'' when monitoring node 'ns=2;s=NZ070_Darfield_NZS070SRV01X.C270.NZ070_B05VA0506_AlmFault' on endpoint 'opc.tcp://10.70.64.95:49320'. Continue.
[04:34:25 ERR] Exception
Opc.Ua.ServiceResultException: Could not send request to server.
   at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest request)
   at Opc.Ua.SessionClient.CreateMonitoredItems(RequestHeader requestHeader, UInt32 subscriptionId, TimestampsToReturn timestampsToReturn, MonitoredItemCreateRequestCollection itemsToCreate, MonitoredItemCreateResultCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Subscription.CreateItems()
   at Opc.Ua.Client.Subscription.ApplyChanges()
   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in /app/opcpublisher/OpcSession.cs:line 561
```

I have tried removing the tags that are reporting errors but it appears to just report the next tag as error'd after this. 

This error is also mentioned on 2 other OPC github projects. 
https://github.com/OPCFoundation/UA-.NET-Legacy/issues/107
https://github.com/open62541/open62541/issues/434

Neither is very clear to me why this might be happening, but sounds like it could be something going over the size limit for an OPCUA message?

markusstuhler, 2019-04-30:
It looks as if you were trying to publish ~60000 nodes. The payload of setting up such a subscription request is too large for the client to encode based on it's limits.

Is it really necessary to monitor that amount of items or can you limit the number to the nodes that are interesting to you? Can you give as more information about you scenario you're using publisher? That would help us define our roadmap.

jmbrunskill, 2019-04-30:
We are setting up a Industrial Historian Solution for a manufacturing business.

The expectation is to be able to support 200k nodes per site (we currently have just 1 IoT Edge per site)

I'm pretty sure we already have a site running with over 100k but perhaps it has to do with the length of the node names? Maybe they are longer at this side. 

If this size is indeed the issue, perhaps the OPC Publisher needs to be able to split the nodes into multiple subscriptions? It looks like all the node ids might be being added to a single subscription at the moment?

jmbrunskill, 2019-04-30:
These logs show a site running with 167568 nodes subscribed, so don't think it is purely a issue with the number of nodes?

The published nodes file for this site is 34 M which is more than the 11M for the site we are having the 'BadEncodingLimitsExceeded' issue one

```
[00:21:48 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://10.60.64.95:49320'.
[00:21:49 INF] Create unsecured session for endpoint URI 'opc.tcp://10.60.64.95:49320' with timeout of 10000 ms.
[00:21:50 INF] Session successfully created with Id ns=1;g=62023d9c-44e7-4508-a296-86c0927353ea.
[00:21:50 INF] The session to endpoint 'opc.tcp://10.60.64.95:49320/' has 3 entries in its namespace array:
[00:21:50 INF] Namespace index 0: http://opcfoundation.org/UA/
[00:21:50 INF] Namespace index 1: urn:NZS060HSTOPC01X.prod01.inroot.net:PTC.ThingWorxIC.1:UA%20Server
[00:21:50 INF] Namespace index 2: ThingWorx Industrial Connectivity
[00:21:50 INF] Created subscription with id 155 on endpoint 'opc.tcp://10.60.64.95:49320'
[00:21:50 INF] Create subscription on endpoint 'opc.tcp://10.60.64.95:49320' requested OPC publishing interval is 30000 ms. (revised: 30000 ms)
[00:21:50 INF] Start monitoring items on endpoint 'opc.tcp://10.60.64.95:49320'. Currently monitoring 0 items.
[00:21:50 INF] Now monitoring 10000 items in subscription with id '155'
[00:21:51 INF] Now monitoring 20000 items in subscription with id '155'
[00:21:51 INF] Now monitoring 30000 items in subscription with id '155'
[00:21:52 INF] Now monitoring 40000 items in subscription with id '155'
[00:21:53 INF] Now monitoring 50000 items in subscription with id '155'
[00:21:54 INF] Now monitoring 60000 items in subscription with id '155'
[00:21:56 INF] Now monitoring 70000 items in subscription with id '155'
[00:21:56 INF] Now monitoring 80000 items in subscription with id '155'
[00:21:57 INF] Now monitoring 90000 items in subscription with id '155'
[00:21:58 INF] Now monitoring 100000 items in subscription with id '155'
[00:21:59 INF] Now monitoring 110000 items in subscription with id '155'
[00:22:00 INF] Now monitoring 120000 items in subscription with id '155'
[00:22:01 INF] Now monitoring 130000 items in subscription with id '155'
[00:22:02 INF] Now monitoring 140000 items in subscription with id '155'
[00:22:03 INF] Now monitoring 150000 items in subscription with id '155'
[00:22:04 INF] Now monitoring 160000 items in subscription with id '155'
[00:22:07 INF] Done processing unmonitored items on endpoint 'opc.tcp://10.60.64.95:49320' took 17247 msec. Now monitoring 167568 items in subscription with id '155'.
```

jmbrunskill, 2019-05-01:
It turns out there was a typo in the node path which was causing this error. 

@markusstuhler  Do you think if all 60k node id's were returning an error status that could be what was causing the encoding limits to be hit?

Is there any plans to add a feature where opcpublisher can report errors or OPC Status for configured nodes? Currently if we have miss configuration it's hard to know why it is not working. 

I'm sure the OPC UA protocol must have the ability to return a status or quality for a node? Is there a debugging flag we could turn on to report this or something?

markusstuhler, 2019-05-02:
Hi @jmbrunskill, could you share more information concerning this issue? What did you change so that it worked again? I'd like to reproduce this on my side.

jmbrunskill, 2019-05-03:
Hi @markusstuhler,

All the node Ids we had configured had path something like this:
'ns=2;s=NZ070_Darfield_NZS070SRV01X.C270.NZ070_B05VA0506_AlmFault'
It turned out this should have been
'ns=2;s=NZ070_Darfield_NZS070**SVR**01X.C270.NZ070_B05VA0506_AlmFault'

I guess to recreate this you'd perhaps need to create a large list of nodeids and subscribe them all with an incorrect node name?

We are using ThingWorx Industrial Connectivity 8.4.66 server as out OPC UA server. So it is possible this treats invalid path's differently to out OPC UA servers.

markusstuhler, 2019-08-01:
Just to clarify, can you confirm that your scenario now works after your change?

Thanks!
Markus