Azure/iot-edge-opc-publisher-v1 #238: opc publisher 2.3.0 fails to connect to edgeHub of edge device

JuergenKosel, 2019-05-28:
On one of our edge devices we have the problem, that the publisher fails to connect to the edgeHub.
The log of the publisher container contains the following
```
[12:34:51 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)

[12:34:51 INF] Trusted issuer store contains 0 certs

[12:34:51 INF] Trusted issuer store has 0 CRLs.

[12:34:51 INF] Trusted peer store contains 1 certs

[12:34:51 INF] 01: Subject 'CN=publisher' (thumbprint: EC9D5A9FD6AC90B720945C90CE8D6F299D7E838A)

[12:34:51 INF] Trusted peer store has 0 CRLs.

[12:34:51 INF] Rejected certificate store contains 0 certs

[12:34:51 INF] There is no site configured.

[12:34:51 INF] Starting server on endpoint opc.tcp://publisher:62222/UA/Publisher ...

[12:34:52 INF] Server started.

[12:34:52 INF] Using default telemetry configuration.

[12:34:52 INF] The name of the configuration file for published nodes is: ./pn.json

[12:34:52 INF] Attemtping to load node configuration from: ./pn.json

[12:34:52 INF] Loaded 0 config file entry/entries.

[12:34:52 INF] There are 0 nodes to publish.

[12:34:52 INF] Create IoTEdgeHub module client using 'Mqtt_Tcp_Only' for communication.

[12:34:53 INF] IoTCentral mode: False

[12:34:53 INF] Certificate 'CN=python-opcua.example.org' will be trusted, because of corresponding command line option.

[12:34:53 ERR] Failure initializing hub communication processing.

System.Security.Authentication.AuthenticationException: TLS authentication error. ---> System.AggregateException: One or more errors occurred. (The remote certificate is invalid according to the validation procedure.) ---> System.Security.Authentication.AuthenticationException: The remote certificate is invalid according to the validation procedure.

   at System.Net.Security.SslState.StartSendAuthResetSignal(ProtocolToken message, AsyncProtocolRequest asyncRequest, ExceptionDispatchInfo exception)

   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)

   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)

   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)

   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)

   at System.Net.Security.SslState.PartialFrameCallback(AsyncProtocolRequest asyncRequest)

--- End of stack trace from previous location where exception was thrown ---

   at System.Net.Security.SslState.ThrowIfExceptional()

   at System.Net.Security.SslState.InternalEndProcessAuthentication(LazyAsyncResult lazyResult)

   at System.Net.Security.SslState.EndProcessAuthentication(IAsyncResult result)

   at System.Net.Security.SslStream.EndAuthenticateAsClient(IAsyncResult asyncResult)

   at System.Net.Security.SslStream.<>c.<AuthenticateAsClientAsync>b__46_2(IAsyncResult iar)

   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)

   --- End of inner exception stack trace ---

   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)

   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)

   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)

   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass22_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()

--- End of stack trace from previous location where exception was thrown ---

   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)

   --- End of inner exception stack trace ---

   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)

   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass35_0.<<OpenAsyncInternal>b__0>d.MoveNext()

--- End of stack trace from previous location where exception was thrown ---

   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)

   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()

   at OpcPublisher.HubCommunication.InitHubCommunicationAsync(TransportType transportType) in /app/opcpublisher/HubCommunication.cs:line 130
```

quodt, 2019-07-12:
I'm having a similar issue. I get more or less the same exception also in version `2.2.0`, `2.3.0` and `2.3.3`. 
It looks like the issue is not related to the used protocol because in `2.3.3.` the IoT EdgeHub communication is'Amqp_Tcp_Only' is used. 

What i've tried to fix this:

- tried different versions of the publisher
- re-installed iotedge
- re-created the device in the azure portal (got new connection string)
- deleted the folders `/var/lib/iotedge/hsm/` and `/var/lib/iotedge/cache/`

Unfortunately i was not able to fix this yet.

quodt, 2019-07-12:
I found the problem in my case: I was using a hostname in the iotedge config `/etc/iotedge/config.yaml` which contained an underscore. By altering that to a hostname without the underscore the publisher was working.

markusstuhler, 2019-08-01:
Hi @JuergenKosel,

There were updates in the IoT Hub SDK that fixed issues when connecting to Edge Hub. The updated SDK is included in the latest version of the publisher. Could you try it with that version again?

Thanks
Markus