Azure/iot-edge-opc-publisher-v1 #245: Cannot connect to OPCUA server when using appcertstoretype=Directory

anhame, 2019-07-25:
Hello,

when I'm using 'Directory' as for the appcertstoretype, the connection to an OPC UA server cannot be established, even though I've copied the certificates to the correct folders. The following errors are logged by the opc publisher:
2019-07-25 09:17:41.079 +02:00 [DBG] OPC: ***EXCEPTION*** BadTcpInternalError 'Could not send an Open Secure Channel request.'
2019-07-25 09:17:41.086 +02:00 [DBG] OPC: ***EXCEPTION*** BadTcpInternalError 'Error establishing a connection: Could not send an Open Secure Channel request.'

In the OPC UA server itself I'm also getting an error 'Certificate is not trusted.' 

When I change the appcertstoretype to be X509Store and then move the newly created certificate which is at first also rejected by the OPC UA server to its trusted folder, everything works fine.
What could be the reason for this?
I'm currently using opc publisher version 2.3.3

Any help appreciated...
regards Hartmut

markusstuhler, 2019-07-25:
Hi @anhame 
Could you tell me something about your setup in which the publisher is running? (Host OS, IoT Edge/Standalone, Windows/Linux containers, ...)?

Thanks!
Markus

anhame, 2019-07-25:
Hi @markusstuhler,
Host OS is Windows7, opc publisher being used as a standalone application, no IoT edge, no containers.
Thanks, Hartmut

markusstuhler, 2019-08-01:
Hi @anhame

due to a limitation in the .NET Core and the OPC UA stack, the only supported certificate on windows is X509Store.

Thanks!
Markus

closed 2019-08-01 by markusstuhler