Azure/iot-edge-opc-publisher-v1 #250: Limit publishednodes

richterr, 2019-07-30:
### Issue
After generating a publishednodes.json with >1500 configured tags to monitor I get the following Error Message.

[image]

After shortening the list to ~700 tags the error disappears.
What is the limit for the monitored tags? Is it a limitation of the OPC UA Server or the OPC Publisher?

Thanks in advance.

### Environment
Ubuntu 18.04 LTS
OPC Publisher in IoT Edge Runtime (1.0.7)

markusstuhler, 2019-08-01:
Hi @richterr, this error message is coming from the OPC UA Server, not from the publisher.

May I ask which server you are using?

Thanks!
Markus

richterr, 2019-08-01:
So there is no limit for tags in the publishednodes.json file?
We are running the OPCPublisher against a Siemens S7-1500 with OPC UA Server activated. What causes this kind of error on the OPC UA Server? Is it a performance restriction or some misconfiguration on the server side?
Thanks in advance.

markusstuhler, 2019-08-01:
We're no S7 experts, but it looks like a configuration setting in the S7 or it hits limits what the server can support. Please check the documentation of the server on how many items are supported.

Thanks!
Markus

closed 2019-08-01 by markusstuhler

richterr, 2019-08-01:
Alright, thanks a lot.