Azure/iot-edge-opc-publisher-v1 #265: Is this the very latest version of OPC Publisher?

FMauroy, 2020-11-03:
Hi, I saw that you recently migrated the code of OPC Publisher to a new repo, this one I presume.
I'm a bit lost now, as I can't be sure this one is the latest one. The doc looks a lot like the one for 2.5.3, which is the one we have used the most (and have slightly customized to our case).
I remember that after 2.5, there was another version, that looked completely different, and was partly compatible with this one (in an IIOT repo). Has it been dropped? Or does it have a separate life now? I had to create a module to translate the messages it outputs, as it didn't support message shaping anymore (through the "telemetryConfiguration.json" file). Should I keep this module? We haven't used it in prod yet.
Thanks for any clarification on this.
Regards
fred

tgrusendorf, 2020-11-03:
I'm also suffering from version confusion.  If I pull the latest docker image it is 2.5.3, but I see lots of references on the main page of this repo to functionality in versions 2.6 and 2.7.  I'm dealing with heartbeat issues in 2.5.3 that appear to be resolved in a subsequent version.  Is 2.5.3 still the recommended release for production?

barnstee, 2020-11-09:
We are working on a new OPC Publisher version that combines the features from 2.5.3 (the latest production version from this repo) with the features from 2.7.199 (the latest production version of the OPC Publisher from the IIoT repo). Release date is TBD, we will inform folks through the main readme.md in this repo as soon as we have planned it.
Regarding which version you should use for production, it depends on the features you need, both a production ready. Once the new version is released, you won't need to chose.
Incidentally, can you send me the format of the message you have shaped?

buzzfrog, 2020-11-09:
Thanks for that answer.
Does this means that Configuration of the simple JSON telemetry format via Separate Configuration File will be part of this merged OPC Publisher? Or is it forever a removed feature from OPC Publiser?

barnstee, 2020-11-09:
This is exactly what we are trying to determine. We would like to get rid of at least the config file and, if needed, provide the message shaping through other mechanisms. Also, we want to move folks to standardized OPC UA PubSub JSON format, instead of everyone having their own custom format.

MatthijsvdVeer, 2020-11-09:
Does this mean that the configuration through Direct Methods will end up in the IIoT repository? That would be excellent news.

barnstee, 2020-11-09:
No, OPC Publisher will continue to be developed in this repo, but yes, the next version will received IoT Hub direct methods support once again.

buzzfrog, 2020-11-09:
Thanks Erich.
The problem for me as a user of OPC Publisher a still needs to have this configuration file, because I want to be able to configure a transformation of the messages. So the solution could be that I create my own module, driven with it's own configuration, to do the transformation.

If we should talk about SoC, is this maybe the best way to handle this? So not one module have to much responsible.

barnstee, 2020-11-09:
Yes, that would be an option. Can you share the transformation you need to do, to see how we can best support you?

tgrusendorf, 2020-11-09:
Thank you Erich, your responses are very helpful.  I'm glad to see that 2.7.199 is considered production ready, and that IoT Hub direct method support will be returning.  It is important in our application to be able to update the published nodes remotely, preferably without having to punch holes through firewalls to access the REST API or connect to OPC Publisher's OPC server.

barnstee, 2020-11-10:
No (inbound) holes in firewalls required. Are you using the OPC Publisher's OPC Server? Why?

buzzfrog, 2020-11-10:
One thing that we want have is an extensible configuration format in the json file. For example, we want to add more metadata to the tag and doing this without having some naming standard for Displayname. In our case we want to register the Machine where the Tags belong to. This because we split up each machine as it's own device into IoT Hub.

tgrusendorf, 2020-11-10:
I'm not currently using OPC Publisher's OPC Server.  I was planning to look into using it to control published nodes since the direct methods were removed (and also just for curiosity).  I don't have a specific need for it if I can use direct methods.

FMauroy, 2020-11-11:
In our case, we had to disable the internal OPC server as in some cases (undetermined, happened only at a couple customers), the creation of the self-signed certificate failed (the keys it creates do not match?!?) and made publisher fail to start. I reported it, but never got any answer.
For the message shaping, we use a very simple format composed of 3 items: HardwareId (mapped to the displayname), SensorValue and Timestamp. For testing the newer version, I had to implement a translator module that would detect messages in the new format and output messages in the new one. Nothing crazy, just a bunch of lines, but an extra step we didn't need before.
Finally, for the configuration, our customers like to play with the json file, they all end up creating scripts to generate it from their OPC system, or directly editing the file on the edge device, as it is in their network. We started implementing a remote configuration feature, split between OPC Publisher 2.5.3 direct methods for subscribing, and a custom module for listing the subscribed and available nodes (using OPC browse requests). At the time, OPC Twin was not available, and now that we tested it, it's too crazy complicated for just browsing a couple OPC servers ;-)
Hope this helps you see what we use and need, which is always based on our customers' feedback. We'd rather not reinvent the wheel as much as possible.
Thanks!

barnstee, 2020-11-11:
Great feedback, thank you. Totally aligned with what we hear from others. We will simplify the publishing/config/deployment story and create better config tools. ETA TBD, but this is our highest priority.

FMauroy, 2020-11-11:
Great to hear! Looking forward to testing the next version.
Also, please retain the use of the "store & forward" feature of the edge hub, we've had several occurrences where the network had lapses, and the last thing our customers want is having gaps in their cloud data. On one occasion the problem was in the agent or hub side, publisher was pushing data fine (according to its logs), but the edge runtime stopped forwarding them to the iot hub, without any error or warning. Restarting the runtime solved the issue but the data was lost forever. Probably an issue for another team ;-)

barnstee, 2020-11-11:
Yes, please log here: https://github.com/Azure/iot-edge-v1/issues

barnstee, 2020-11-23:
Closing since question was answered.

closed 2020-11-23 by barnstee