Azure/iot-edge-opc-publisher-v1 #266: Dynamic information model and OPC Publisher

fijo1015, 2020-11-05:
Hi everyone,

I am using a dynamic OPC UA information model. What I mean by dynamic is, that in the beginning the information model contains a few nodes and some methods. By calling these methods new nodes will be created. So the information model is growing while using it. 
I'd like to know, if the OPC Publisher Module can handle nodes that are not present in the beginning. In the configuration file (publishednodes.json) i want to list some nodes that are not present in the beginning. Is OPC Publisher able to register the appearing nodes and then publish it to the IoT Hub?

Regards
jo

barnstee, 2020-11-09:
This is currently not supported. However, if the server sends out a "DataModelChanged" event (check the OPC UA spec, not sure if the name is 100% accurate), we could hook into this and try to publish the new nodes then.

buzzfrog, 2020-11-09:
Is it this you mean: ModelChangeEvents are generated to indicate a change of the AddressSpace structure.
https://reference.opcfoundation.org/v104/Core/docs/Part3/9.32.1/

barnstee, 2020-11-09:
Yep, that's it.