Azure/iot-edge-opc-publisher-v1 #268: Q: When running opc-publisher as Iot Edge Module, do you need to configure connection string to Iot Hub?

perttvil, 2020-11-12:
I am running docker container mcr.microsoft.com/iotedge/opc-publisher:2.3.3 as a module inside Iot Edge runtime.
Everything works fine with version 2.3.3. 

It does not have any configuration directly to Iot Hub (no connections string) and communication to Iot Hub goes via Iot Edge runtime routes and internal modules.

If I update mcr.microsoft.com/iotedge/opc-publisher:2.3.3 to any newer version, lets say 2.5.4 it wont start at all and says:

`
OPCPublisher                  | [13:24:50 INF] Attempting to read device connection string from cert store using subject name: publisher
OPCPublisher                  | [13:24:50 FTL] Device connection string not found in secure store. Please pass it in at least once via command line option. Can not connect to IoTHub. Exiting...
OPCPublisher                  | [13:24:50 FTL]    at OpcPublisher.IotHubCommunication..ctor() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\IotHubCommunication.cs:line 157
OPCPublisher                  |    at OpcPublisher.IotHubCommunication.get_Instance() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\IotHubCommunication.cs:line 62
OPCPublisher                  |    at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\Program.cs:line 742
OPCPublisher                  | System.ArgumentException: Device connection string not found in secure store. Please pass it in at least once via command line option. Can not connect to IoTHub. Exiting...
OPCPublisher                  |    at OpcPublisher.IotHubCommunication..ctor() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\IotHubCommunication.cs:line 157
OPCPublisher                  |    at OpcPublisher.IotHubCommunication.get_Instance() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\IotHubCommunication.cs:line 62
OPCPublisher                  |    at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\Program.cs:line 742
OPCPublisher                  | [13:24:50 FTL] Publisher exiting... 
`

So do newer versions really need connection string configuration to Iot Hub and if so, why? I read other old github issues, and from those I understood that connection string configuration should not be necessary when working inside Iot Edge runtime.

So what is the truth?

OpcPubisher module configuration:
`"OPCPublisher": {
                        "settings": {
                            "image": "mcr.microsoft.com/iotedge/opc-publisher:2.5.4",
                            "createOptions": "{\"Hostname\":\"publisher\",\"Cmd\":[\"--pf=/appdata/pn.json\",\"--aa\"],\"HostConfig\":{\"Binds\":[\"/opt/iotedge/opc-ua:/appdata\"],\"ExtraHosts\":[\"mbp-vp.local:192.168.1.107\"]}}"
                        },
                        "type": "docker",
                        "status": "running",
                        "restartPolicy": "never",
                        "version": "1.0"
                    }`

Published nodes configuration (pn.json):
`[
  {
    "EndpointUrl": "opc.tcp://mbp-vp.local:53530/OPCUA/SimulationServer",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=3;i=1001"
      },
      {
        "Id": "ns=3;i=1002"
      }
    ]
  }
]`

perttvil, 2020-11-13:
I noticed that mcr.microsoft.com/iotedge/opc-publisher:2.5.3 works fine with real iotEdge runtime (1.0.10.2).
The problem is when running inside dev runtime (iotedgehubdev, version 0.14.3).

No idea how to fix the problem though.

perttvil, 2020-11-16:
In version 2.3.3 'IsIotEdgeModule' is detected like this (and it works):
https://github.com/Azure/iot-edge-opc-publisher/blob/2.3.3/opcpublisher/IotEdgeHubCommunication.cs#L19

barnstee, 2020-11-23:
Yes, we removed support again in 2.5.4 since we had compatibility issues with the simulation environment. If you are dependent on this, please stay on 2.3.3 for now. We will eval adding this back in in version 3.0 of OPC Publisher.

perttvil, 2020-11-23:
Ok. Any ETA for version 3.0? First of year 2021?

barnstee, 2020-11-23:
Yeah, in Spring. Don't have an exact date yet.