Azure/iot-edge-opc-publisher-v1 #269: Not able to use custom certificate signed by root CA certificate

SandeepThombare, 2020-11-12:
I am using OPC publisher 2.5.4 version with latest version of IOT edge daemon.

I have created the root certificate for signing self signed certificate using below instructions,

**Create root certificate and root key**

1. openssl genrsa -out cakey.pem 2048

2. openssl req -new -key cakey.pem -x509 -days 365 -out cacert.pem

**Create certificate files (.der and .pfx)for OPC publisher using above root certificate and key.**

1. openssl genrsa -out server.pem 2048

2. openssl req -new -key server.key -out server.csr -subj /CN=publisher

3. openssl x509 -req -days 365 -extfile cert.ext -in servercsr -CA cacert.pem -CAkey cakey.pem -CAcreateserial -out server.der

4. openssl pkcs12 -export -keypbe NONE -certpbe NONE -in server.der -inkey server.pem-out server.pfx

I copied .der and .pfx files to /pki/own/certs and /pki/own/private respectively.

My Opc publisher picked up this certificate and sent to KEPServer EX 6 but KEPServer Ex 6 is not accepting the certificate and showing error.
[image]

I have tried to install the certificate chain but still no success and KEPServer 6 is showing same error.

Attaching the certificate for reference.
[certificate files.zip](https://github.com/Azure/iot-edge-opc-publisher/files/5532594/certificate.files.zip)

please let me know if anything is missing here.

mregen, 2020-11-17:
Hi @SandeepThombare, the publisher should have the CA cert as well in pki/issuer/certs as e.g. cacert.der, if it is not there it might cause validation issues because publisher cannot validate his own chain. 
Whats unclear to me (and I have a Kepware server running here too) how did you import a CA cert to the Kepware server? I see only import for the server cert.

cristipogacean, 2020-11-17:
Does this help: https://github.com/Azure/Industrial-IoT/issues/757?

SandeepThombare, 2020-11-18:
@mregen Thanks for your reply.

I tried to test by copying cacert.der to /pki/issuer/certs but still same error. KEPServerEx 6 is not accepting this certificate.it is giving same error mentioned earlier.**Note**:- I am using opc publisher version 2.5.3 for this testing as i am getting some issue with opc publisher version 2.5.4 version on my system. Hope that will not make big difference in this testing.

Please find below details for your reference,
**Container create option,**
{
"Hostname": "publisher",
"Cmd": [
"--pf=/appdata/pn.json",
"--tm=true",
"--aa",
"--ap=/mount/pki/own",
"--tp=/mount/pki/trusted",
"--rp=/mount/pki/rejected",
"--ip=/mount/pki/issuer",
"--lf=/mount/pki/publisher.log",
"--kt=4",
"--si=38",
"--ll=debug"
],
"HostConfig": {
"Binds": [
"/home/iiot/certscript/nov9:/appdata",
"/home/iiot/certscript/nov10:/mount"
]
}
}

**Publisher Log**
IoTEdge detected.
[07:56:23 INF] Current directory is: /app
[07:56:24 INF] Log file is: /mount/pki/publisher.log
[07:56:24 INF] Log level is: debug
[07:56:24 INF] OPC Publisher V2.5.3 starting up...
[07:56:24 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[07:56:24 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[07:56:24 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[07:56:24 INF] opcstacktracemask set to: 0x3A4
[07:56:24 INF] OperationTimeout set to 120000
[07:56:24 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[07:56:24 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[07:56:24 INF] Trusted Issuer store type is: Directory
[07:56:24 INF] Trusted Issuer Certificate store path is: /mount/pki/issuer
[07:56:24 INF] Trusted Peer Certificate store type is: Directory
[07:56:24 INF] Trusted Peer Certificate store path is: /mount/pki/trusted
[07:56:24 INF] Rejected certificate store type is: Directory
[07:56:24 INF] Rejected Certificate store path is: /mount/pki/rejected
[07:56:24 INF] Rejection of SHA1 signed certificates is disabled
[07:56:24 INF] Minimum certificate key size set to 1024
[07:56:24 INF] Application Certificate store type is: Directory
[07:56:24 INF] Application Certificate store path is: /mount/pki/own
[07:56:24 INF] Application Certificate subject name is: publisher
[07:56:24 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[07:56:24 INF] Application certificate with thumbprint '403027F521EBF1A47D8319D424D6344728CA9346' found in the application certificate store.
[07:56:24 INF] Application certificate is for ApplicationUri 'urn:publisher:publisher:microsoft:', ApplicationName 'publisher' and Subject is 'publisher'
[07:56:24 INF] Adding server certificate to trusted peer store. StorePath=/mount/pki/trusted
[07:56:24 WRN] Can not add server certificate to trusted peer store. Maybe it is already there.
System.ArgumentException: A certificate with the same thumbprint is already in the store.
   at Opc.Ua.DirectoryCertificateStore.Add(X509Certificate2 certificate, String password)
   at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\OpcApplicationConfigurationSecurity.cs:line 240
[07:56:24 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
[07:56:24 INF] Trusted issuer store contains 1 certs
[07:56:24 INF] 01: Subject 'CN=CARoot, O=Retail, S=Mum, C=BE' (thumbprint: 2553F812C9F89B21EEAAE2CE16E10DA9F35BAABB)
[07:56:24 INF] Trusted issuer store has 0 CRLs.
[07:56:24 INF] Trusted peer store contains 1 certs
[07:56:24 INF] 01: Subject 'CN=publisher' (thumbprint: 403027F521EBF1A47D8319D424D6344728CA9346)
[07:56:24 INF] Trusted peer store has 0 CRLs.
[07:56:24 INF] Rejected certificate store contains 0 certs
[07:56:24 INF] There is no site configured.
[07:56:24 INF] Starting server on endpoint opc.tcp://publisher:62222/UA/Publisher ...
[07:56:25 DBG] OPC: MasterNodeManager.Startup - NodeManagers=3
[07:56:26 INF] Server started.
[07:56:26 INF] Using default telemetry configuration.
[07:56:26 INF] The name of the configuration file for published nodes is: /appdata/pn.json
[07:56:26 INF] Attemtping to load node configuration from: /appdata/pn.json
[07:56:26 INF] Loaded 1 config file entry/entries.
[07:56:26 INF] There are 3 nodes to publish.
[07:56:26 INF] Create module client using 'Amqp_Tcp_Only' for communication.
[07:56:26 INF] IoTCentral mode: False
[07:56:26 DBG] Open hub communication
[07:56:32 INF] Connection status changed to 'Connected', reason 'Connection_Ok'
[07:56:32 DBG] Register desired properties and method callbacks
[07:56:32 DBG] Init D2C message processing
[07:56:32 INF] Message processing and hub communication configured with a send interval of 38 sec and a message buffer size of 262144 bytes.
[07:56:32 INF] Creating task process and batch monitored item data updates...
[07:56:32 INF]
[07:56:32 INF]
[07:56:32 INF] Publisher is running. Press CTRL-C to quit.
[07:56:32 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://publisherXXX'.

Regarding, importing Cacert.der, I have imported root certificate to trusted root certificate authority of my windows machine. by doing this i can see certificate path gets updated,
[image]

@cristipogacean , It worked for version 2.5.7. there was difference in certificate creation procedure. In that case I was not using any CA root certificate to sign the server.der certificate. 
As per our last findings ,I am using cert.ext with below configuration

authorityKeyIdentifier=keyid,issuer
basicConstraints=CA:TRUE
keyUsage=digitalSignature, nonRepudiation, keyEncipherment, dataEncipherment, keyCertSign, cRLSign
extendedKeyUsage=clientAuth
subjectKeyIdentifier=hash
subjectAltName=@alt_names

[alt_names]
URI.1 = urn:publisher:publisher:microsoft:
DNS.1 = publisher

Please let me know if any further changes required.

mregen, 2020-11-19:
Hi @SandeepThombare, regarding: 

> importing Cacert.der, I have imported root certificate to trusted root certificate authority of my windows machine.

That's not where you import the root CA. You need to import the root CA to the trusted store of the KepWare server, here:

[image]

The explanation is, if a OPC server were to trust all certificates issued by the root CA's in the windows trusted root store, anybody with such a signed cert could connect. That's why OPC servers have a separate trusted store. 

I tested with a root cert created by OPC Vault and KEPWARE server accepts the secure connection after import. For reference I attached a cert and the CA issued for your publisher. Technically I believe your CA cert should be accepted once the CA is imported at the right place in the KEP Ware server.

[publisher and CA cert.zip](https://github.com/Azure/iot-edge-opc-publisher/files/5566199/publisher.and.CA.cert.zip)

mregen, 2020-11-27:
@SandeepThombare please reopen if issue is still not solved...

closed 2020-11-27 by mregen