Azure/iot-edge-opc-publisher-v1 #270: Incorrect timestamps using the "SkipFirst" and "HeartbeatInterval" features

fijo1015, 2020-11-13:
Hi everyone,

during my tests i noticed that the "SkipFirst" (set to false) and the "HeartbeatInterval" is not correct every time. The SourceTimestamps are incorrect by exactly one hour. The following examples will show you the issue:

1.) SkipFirst:
last value:  "Value":{"Value":0,"SourceTimestamp":"2020-11-11T13:03:06.1880000+01:00"}
--> Start OPC Publisher using "SkipFirst" set to "False"
next value: "Value":{"Value":0,"SourceTimestamp":"2020-11-11T12:06:51.4060000Z"}

2.) HeartbeatInterval (set to 60s):
last value:   "Value":{"Value":-2.01558608281E-05,"SourceTimestamp":"2020-11-13T15:35:37.9750000+01:00"}
next value: "Value":{"Value":-2.01558608281E-05,"SourceTimestamp":"2020-11-13T14:36:37.9750000Z"}

How can I fix this issue?

Best regards
jo

perttvil, 2020-11-13:
Are u sure there is an issue?
First timestamp timezone is +01:100 and second Utc.

fijo1015, 2020-11-16:
Maybe i misunderstand something. I wonder why there is a usage of different timezones at the same application and data source.

koepalex, 2020-11-16:
The SourceTimestamp is generated by the OPC UA Server the OPC Publisher is connected to. 

The field that is containing the timestamp also provides the time zone information (`+01:00`, `Z`). With this timezone information it can be processed by telemetry processor/timeseries database. For example when you use write the teleemetry processor in C# you can use the DateTime class from .NET framework and just load the timestamps, comparission, ordering etc. will work without extra work. If you don't want to handle/store different time zones you can use the DateTime to get the timestamp as UTC.

I'll close the issue, if something is still unclear please reopen.

closed 2020-11-16 by koepalex