Azure/iot-edge-opc-publisher-v1 #271: OpcPublisher: BadSubscriptionIdInvalid

fijo1015, 2020-11-16:
Hi,

We are testing OPC Publisher since a few days. If we choose a publishing interval around 200ms we get an BadSubscriptionIdInvalid-Error after a few seconds/minutes and the subscription gets lost. 

**error message**
2020-11-16 16:27:10.282 +01:00 [ERR] Exception
Opc.Ua.ServiceResultException: BadSubscriptionIdInvalid
   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled) in C:\Build\Projects\NETStandard-NuGet\src\SampleApplications\SDK\Opc.Ua.Client\Subscription.cs:line 1069
   at OpcPublisher.OpcUaSubscription.SetPublishingMode(Boolean enabled) in D:\dev\IoT_Publisher\opcpublisher\OpcUaSubscription.cs:line 76
   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in D:\dev\IoT_Publisher\opcpublisher\OpcSession.cs:line 731

**OS**
Windows 10 Enterprise

Is that an OPC Publisher issue?

best regards
jo

fijo1015, 2020-11-17:
I forgot to mention the version: We are using v2.4.1

monikavar, 2020-11-17:
This is fixed in v 2.5.3. Please retest. Thanks!

closed 2020-11-17 by monikavar

fijo1015, 2020-11-17:
Do you mean v2.5.3 of the IIoT repo?