Azure/iot-edge-opc-publisher-v1 #275: opc publisher error

bellenkurt, 2020-11-26:
Hello we get this error in the output of our opc module

[16:09:03 VRB] OPC: Bytes written: 106
[16:09:03 DBG]    ApplicationUri: urn:BEHQWINFTPRD01:FactoryTalk Linx Gateway OPC UA Server
[16:09:03 DBG]    EndpointUrl: opc.tcp://behqwinftprd01:4990/behqwinftprd01
[16:09:03 DBG]    DisplayName: ns=3;s=DS/RSLE::[PWM002]F21:2
[16:09:03 VRB] OPC: PUBLISH #208 SENT
[16:09:03 DBG]    Value: 458.367676
[16:09:03 DBG] Enqueue a new message from subscription 3428698689
[16:09:03 DBG]  with publishing interval: 0 and sampling interval: 1000):
[16:09:03 ERR] Error processing monitored item notification
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpcPublisher.OpcMonitoredItem.MonitoredItemNotificationEventHandler(MonitoredItem monitoredItem, MonitoredItemNotificationEventArgs e) in D:\a\1\s\modules\src\Microsoft.Azure.IIoT.Modules.OpcUa.Publisher\src\OpcMonitoredItem.cs:line 566

this is the config of the container create options

{
    "Hostname": "publisher",
    "Cmd": [
        "--pf=/appdata/publishednodes.json",
        "--aa",
        "--ms=0",
        "--si=10",
        "--ll=verbose"
    ],
    "HostConfig": {
        "Binds": [
            "c:/iiotedge:c:/appdata"
        ]
    }
}

this is the config of hte publishednodes.json

[
  {
    "EndpointUrl": "opc.tcp://behqwinftprd01:4990/behqwinftprd01",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=3;s=DS/RSLE::[PWM002]F21:2"
      }
    ]
  }
]

does anyone has any idea why we get this error

barnstee, 2020-11-27:
Note: This happens with OPC Publisher 2.5.4.

bellenkurt, 2020-11-27:
what version should we the run, and can you maybe give us the command line to configure this; this is the command line we run now for the current module

mcr.microsoft.com/iotedge/opc-publisher:latest

bellenkurt, 2020-11-27:
> Note: This happens with OPC Publisher 2.5.4.

what version should we the run, and can you maybe give us the command line to configure this; this is the command line we run now for the current module

mcr.microsoft.com/iotedge/opc-publisher:latest

barnstee, 2020-11-27:
Please use 2.5.3-debug and send me the new verbose log, please.

bellenkurt, 2020-11-27:
mcr.microsoft.com/iotedge/opc-publisher:2.5.3-debug

like this?

barnstee, 2020-11-27:
yep

bellenkurt, 2020-11-27:
[publishlog.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5608465/publishlog.txt)
this is the updated log with version 2.5.3 => i still see that error

barnstee, 2020-11-27:
With the debug output, we could analyze the issue and it is a rare race condition in the OPC Publisher, which we fixed in subsequent versions. Please try version 2.7.199-debug.

barnstee, 2020-11-27:
Also, in some cases, the error is fixed by restarting IoT Edge. Please try this before going to version 2.7.

bellenkurt, 2020-11-30:
hello Erich

thank you very much for the help, this solved the issue; just a reboot and restart from the publisher module on the edge device together with the 2.5.3-debug version

thx
kurt

barnstee, 2020-11-30:
Ok, good. For performance, feel free to switch back to an optimized/release build, i.e. without the -debug.

closed 2020-11-30 by barnstee