Azure/iot-edge-opc-publisher-v1 #280: error "Could not send worker heartbeat"

franperezlopez, 2020-12-01:
Hi,

I've just update my iotedge module, which is built using the image mcr.microsoft.com/iotedge/opc-publisher:latest, and when is deployed in the iotedge device, start failing and doesn't collect any data ... is there any change I need to be aware??

[image]

monikavar, 2020-12-02:
What is the setup and which version of the OPC Publisher were you using before?

Additionally, are you using industrial IoT microservices?

barnstee, 2020-12-09:
Please get back to our question, or is the issue resolved?

barnstee, 2020-12-09:
@franperezlopez

franperezlopez, 2020-12-11:
Sorry for the late response,

a) I did additional tests, and the problem seems not related to the opcpublisher version. There is a telegraf module (docker) for ingesting data from opcserver directly to influxdb, and I get this error when opcpublisher and telegraf are working at the same time. if opc publisher is working alone, it works fine
b) I don't know what you mean by Industrial IoT microservices ... how can I check that?

barnstee, 2020-12-15:
@franperezlopez , if you are not using industrial IoT services, it explains the error: The worker heartbeat is something the OPC Publisher is trying to send to the industrial IoT services. If you are using a publishesnodes.json file, the OPC Publisher should automatically switch to standalone more, not requiring the industrial IoT services. Do you have one configured?
Not sure how the telegraph modules is affecting this, it shouldn't as each module is running in its own container.

franperezlopez, 2020-12-17:
there was an error in the configuration of the deployment template, so no configuration file was set. a little bit "bizarre" that because the json file is missing, I get an error of the heartbeat, a warning message would have been more helpful .. but whatever, now is working. thanks for your support

closed 2020-12-17 by franperezlopez

sagarganachari, 2021-01-04:
> there was an error in the configuration of the deployment template, so no configuration file was set. a little bit "bizarre" that because the json file is missing, I get an error of the heartbeat, a [...]

How did you resolve this issue @franperezlopez 

https://github.com/Azure/Industrial-IoT/issues/993

franperezlopez, 2021-01-04:
@sagarganachari  .. this error was provoked by a typo in my deployment template. 

check your published docker in the iotedge device, then open a bash terminal into the running docker (`docker exec ... bash`), and check that the configuration file which you provide as `createOptions` (for example, `--pf=/app/opcserver.json` or whatever you have) actually exists.

sagarganachari, 2021-01-04:
> pen a bash terminal into the running docker

@franperezlopez  i checked container option path /app/publishenodes.json exists in linux edge device.

iotedge status showing OPCPublisher running.....
[image]

franperezlopez, 2021-01-04:
in my case, it didn't matter .. the opc publisher might be running, but the config file might be missing ... as I said in my comments before, the opc publisher does not display an error message if the config file is missing. 

if you say that the config file it is on the edge device, that means that you need to bind that volume to the docker, right? again, open a bash console against the running opcpublisher, and check that you can execute a command like `ls /app/publishenodes.json` to make sure opcpublisher has access to the config file. otherwise, you're hitting a different problem that was mine.

sagarganachari, 2021-01-04:
@franperezlopez @monikavar 

Unfortunately yes i am hitting a different problem

after executing the command i am accessing the path...find screen shot

[image]

barnstee, 2021-01-04:
You just ran that command on the host, you need to run the command *within* the container. That's what @franperezlopez meant with "against the container".

sagarganachari, 2021-01-04:
> You just ran that command on the host, you need to run the command _within_ the container. That's what @franperezlopez meant with "against the container".
@barnstee 

Yes i ran it within container i am getting ls: cannot access /app/publishenodes.json, attached screen shot

[image]

barnstee, 2021-01-04:
"No such file or directory" -> There's your problem! Please check the directory mapping to the host.

sagarganachari, 2021-01-04:
[image]
[image]
@barnstee Please find the path screenshots ,directory mapping is that same you are talking about....

barnstee, 2021-01-04:
"Binds" in your case should be ["/app:/app"], as you placed your publishesnodes.json in the /app folder on both the host and the container.

sagarganachari, 2021-01-05:
I copied the host file inside container and checked the same .but still i get heartbeat model error.find attached screen shots

to copy the host publishenodes.json file to container   /iiotedge/ location i used the command.

$ tar -cv * | docker exec -i OPCPublisher tar x -C /

Is it right to copy ?? after copy i can  access the file inside container.

[image]
![image](https://
[image]
user-images.githubusercontent.com/76048188/103625146-7849e580-4f32-11eb-8160-a989a75452d1.png

barnstee, 2021-01-05:
No, copying is not the right solution as the file will be gone when the container restarts. You need to map it as @franperezlopez and I described above. Please read up on principles of Docker containers to get an overview of how they work.

sagarganachari, 2021-01-05:
> No, copying is not the right solution as the file will be gone when the container restarts. You need to map it as @franperezlopez and I described above. Please read up on principles of Docker containers to get an overview of how they work.
[image]

@barnstee  After inspecting the container i can see that volume is mounted on the container,,,,,,but still result remains same
[image]

that i cannot access the file with ls /iiotedge/publishenodes.json......