Azure/iot-edge-opc-publisher-v1 #283: User is unable to connect from opc publisher to opcua server from aspentech. User Can connect to opcua server from generic client

daelsala, 2020-12-10:
Below is the Users JSON configurations along with the error received when attempting to connect:

JSON Config:
`json config

[
  {
    "EndpointUrl": "opc.tcp://PENPPRC23:63500/InfoPlus21/OpcUa/Server",
    "OpcAuthenticationMode": "UsernamePassword",
    "OpcAuthenticationUsername": "domain\\userID",
    "OpcAuthenticationPassword": "password",
    "UseSecurity": true,
    "OpcNodes": [
      {
        "Id": "ns=2;i=2186",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 5000,
        "DisplayName": "hmddp"
      }
    ]
  }
]
`

Powershell Output:
`Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS C:\Windows\system32> docker -H npipe:////./pipe/iotedge_moby_engine logs -f OPCPublisher --tail 100
  # Ingress DataChanges (from OPC)     :              0
  # Ingress ValueChanges (from OPC)    :              0
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              0
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              0
  # Encoder avg Notifications/Message  :              0
  # Encoder avg IoT Message body size  :              0 (0%)
  # Encoder avg IoT Chunk (4 KB) usage :              0
  # Estimated IoT Chunks (4 KB) per day:              0
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :              0
  # Connection retries                 :             36

[10:10:35 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:10:45 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:10:55 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:11:05 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:11:15 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:11:25 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session '8e2de58a172f45939359ae6dc072905aa4561010' due to One or more errors occurred. (Error establishing a connection: BadNotConnected)
[10:11:33 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://PENPPRC23:63500/InfoPlus21/OpcUa/Server_8e2de58a172f45939359ae6dc072905aa4561010
  # Ingestion duration                 :    00:00:00:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              0
  # Ingress ValueChanges (from OPC)    :              0
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              0
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              0
  # Encoder avg Notifications/Message  :              0
  # Encoder avg IoT Message body size  :              0 (0%)
  # Encoder avg IoT Chunk (4 KB) usage :              0
  # Estimated IoT Chunks (4 KB) per day:              0
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :              0
[... 37 lines cut ...]
   at System.Threading.ThreadPoolBoundHandleOverlapped.CompletionCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* nativeOverlapped)
   at System.Threading._IOCompletionCallback.PerformIOCompletionCallback(UInt32 errorCode, UInt32 numBytes, NativeOverlapped* pNativeOverlapped)
[10:12:34 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://PENPPRC23:63500/InfoPlus21/OpcUa/Server_8e2de58a172f45939359ae6dc072905aa4561010
  # Ingestion duration                 :    00:00:00:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :              0
  # Ingress ValueChanges (from OPC)    :              0
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :              0
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :              0
  # Encoder avg Notifications/Message  :              0
  # Encoder avg IoT Message body size  :              0 (0%)
  # Encoder avg IoT Chunk (4 KB) usage :              0
  # Estimated IoT Chunks (4 KB) per day:              0
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :              0
  # Connection retries                 :             48

PS C:\Windows\system32>

`

daelsala, 2020-12-10:
@barnstee if you can please assist as soon as possible.

koepalex, 2020-12-11:
The crash is caused by invalid escape sequence, please try `domain\\userID` for "OpcAuthenticationUsername". 
Can you verify if this works?

daelsala, 2020-12-11:
> The crash is caused by invalid escape sequence, please try `"domain\\userID"` for "OpcAuthenticationUsername".
> Can you verify if this works?

@koepalex 
They can push the keys but don’t know how to confirm if it is working correctly.
Do they need to stop/restart the services.
Is there an output command I can run to provide feedback?

koepalex, 2020-12-11:
Restart the publisher: `iotedge restart <publisher-module-name>`
Show the logs: `iotedge logs <publisher-module-name>`

daelsala, 2020-12-11:
> Restart the publisher: `iotedge restart <publisher-module-name>`
> Show the logs: `iotedge logs <publisher-module-name>`

after customer tested the run:
PS C:\Windows\system32> net stop iotedge
...
The Azure IoT Edge Daemon service was stopped successfully.

PS C:\Windows\system32> net start iotedge
The Azure IoT Edge Daemon service is starting.
The Azure IoT Edge Daemon service was started successfully.

daelsala, 2020-12-11:
[OPCTroubleshooting.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5680475/OPCTroubleshooting.txt)

Attached is the log output from cx

daelsala, 2020-12-11:
> [OPCTroubleshooting.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5680475/OPCTroubleshooting.txt)
> 
> Attached is the log output from cx

Contains logs for edgeAgent and edgeHub

daelsala, 2020-12-11:
Latest Logs after restarting attacehd below
[OPCTroubleShooting2.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5680579/OPCTroubleShooting2.txt)

daelsala, 2020-12-11:
Attached are verbose logging files from cx
[OPCTroubleShooting3.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5681723/OPCTroubleShooting3.txt)
[OPCTroubleShooting4.txt](https://github.com/Azure/iot-edge-opc-publisher/files/5681724/OPCTroubleShooting4.txt)

koepalex, 2020-12-14:
`Failed to connect to opc.tcp://penpapp23.apm.ascend.com:63500/InfoPlus21/OpcUa/Server: Error establishing a connection: BadNotConnected - try again...` 

This indicates that the OPC Publisher is not able to connect to OPC UA Server, it is not related to security settings, otherwise there would be an `SecurityCheckFailed` in the log.

koepalex, 2020-12-14:
How to check the connection to OPC UA server from iot edge host system: `Test-NetConnection -ComputerName "PENPPRC23" -Port 63500 -InformationLevel "Detailed"`

koepalex, 2020-12-14:
How to check the connection to OPC UA server from insight OPC Publisher docker container:

`sudo docker exec -it -u root --privileged <publisher-module-name> bash`

```bash
# Install netcat if not available
sudo apt update
sudo apt install netcat
# Test TCP connection to address
nc -z -v PENPPRC23 63500
```

daelsala, 2020-12-14:
Closing issue. Not a bug with OPC publisher, root cause was an unrelated connectivity issue

closed 2020-12-14 by daelsala