Azure/iot-edge-opc-publisher-v1 #285: ARM32V7: Not supported anymore

dvanlaerhoven, 2021-01-07:
Hi All!

I've used the OPCPublisher module this summer with succes on an arm32v7 device. 
I believe I've done the same thing, but now I get:

```
root@axcf2152:/opt/plcnext/# docker run -it mcr.microsoft.com/iotedge/opc-publisher:latest
standard_init_linux.go:211: exec user process caused "exec format error"
failed to resize tty, using default size
``` 

Is arm32v7 not longer supported? I could not find a decapitated message.
Can I build the image myself, what base image should I take?

With Kinds regards

Dries

monikavar, 2021-01-08:
Sorry, arm was never officially supported and should not have been released.

hbunjes, 2021-02-23:
According to the Marketplace ARM is still officially supported: https://azuremarketplace.microsoft.com/en-us/marketplace/apps/microsoft_iot.iotedge-opc-publisher