Azure/iot-edge-opc-publisher-v1 #286: OPCPublisher module was running before but do not run after stop and start

subbu-ai, 2021-01-13:
I am running 8 iotedge OPCPublisher modules (latest version) in a Windows 2019 server using windows docker containers.  6 modules are running good. 2 could not start.

To find out the issue,  I tried to stop one of the running OPCPublisher module by setting the Desired Status to “stopped” using Set Modules menu on the device.  The OPCPublisher module stopped with a failure message.  When I again set the Desired Status to “running”, this module throwing error saying unable to start metric server.  

[21:45:44 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://10.0.3.106:49310_cf9df0a198e2a4285a8571bb30a5d3b70493d245
  ** Ingestion duration                 :    00:06:57:57 (dd:hh:mm:ss)
  Ingress DataChanges (from OPC)     :              1 (0/s)
  Ingress ValueChanges (from OPC)    :             49 (0/s)
  Ingress BatchBlock buffer size     :              0
  Encoding Block input/output size   :              0 | 0
  Encoder Notifications processed    :             49
  Encoder Notifications dropped      :              0
  Encoder IoT Messages processed     :              1
  Encoder avg Notifications/Message  :             49
  Encoder avg IoT Message body size  :         17,011 (6%)
  Encoder avg IoT Chunk (4 KB) usage :            4.2
  Estimated IoT Chunks (4 KB) per day:              0
  Outgress Batch Block buffer size   :              0
  Outgress input buffer count        :              0
  Outgress input buffer dropped      :              0
  Outgress IoT message count         :              0
  Connection retries                 :              **0****

[21:47:36 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to devcgedge01 using MqttOverTcp.
[21:47:36 INF Root] Starting module OpcPublisher version 2.7.206.36449.
[21:47:36 INF Root] Initiating prometheus at port 9702/metrics
[21:47:36 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection
System.Net.HttpListenerException (5): Access is denied.
   at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23

Previously this module was running.  Now it could not start .I did not make any changes, what is the issue?  How to fix this issue?  I am unable to find any documentation on the edge metrics collection or configurations related to that. 

I created a new OPCPublisher module, that also has same problem.   I would really appreciate your help here.

barnstee, 2021-01-14:
Hi @subbu-ai, what does "setting the Desired Status to “stopped” using Set Modules menu on the device" refer to? Is this some app that came with the gateway device? My hunch is that this is not supported by OPC Publisher. OPC Publisher should be restarted using the iotedge command line, i.e. "iotedge restart OpcPublisher".

barnstee, 2021-01-14:
From the separate internal support ticket that was created for this issue, I can see that you are running several OPC Publisher instances on the SAME IoT Edge gateway. This is not supported. Please use a single OPC Publisher instance

subbu-ai, 2021-01-14:
Hi Erich,

This is not a gateway device. IoT Edge Modules (OPCPublisher) running in a windows server on premise.

"setting the Desired Status to “stopped” using Set Modules menu on the device" -  This refers to in Azure Portal,  when you click on the Device ID, then you see all the OPCPublisher images that are running.  Then click on the “Set moddules”.  Then click on the  IoT Edge Module (OPCPublisher).  Then set the desired state to “Stopped”.  Create the edge module. Then again set the desired state to “Running”. Create the edge module.

We used both ways, restart at the edge using command line in the edge server as well as from Azure portal as described above.

Thanks,
Subbu

barnstee, 2021-01-14:
Got it, thank you. Yes, using the portal should work and is equivalent to running iotedge restart <module>. That leaves the issue with the multiple instances on the same gateway. Please use a single instance.

subbu-ai, 2021-01-14:
Please confirm the following:

edgeHub_1 + edgeAgent_1 + OPCPublisher_1
edgeHub_2 + edgeAgent_2 + OPCPublisher_3
…
…

Is this configuration correct? Can all of them run in a single Windows Server?

Thanks,

barnstee, 2021-01-14:
No, please use a single IoT Edge Hub, agent and OPC Publisher. We have tested OPC Publisher with up to 70,000 published tags running on a single instance, the bottleneck is always the internet connection.

subbu-ai, 2021-01-14:
We have 200 (2 seconds) tags, 400 (5 seconds) tags, 6600 (30 seconds)  = total 7200 tags . Can we put all of them in one OPCPublisher + edgeHub + edgeAgent?  Data Source is single OPC/UA tunnel but multiple devices.

barnstee, 2021-01-14:
Yes, you can.

subbu-ai, 2021-01-14:
Thanks.  As we do not want to resample and maintain original sampling provided by the OPC/UA tunnel, we are not mentioning any published_nodes.json file.  Also we have data compression involved in the Pi Server, so it takes care of the heartbeat interval.  So, we do not specify any heartbeat in the published_nodes.json.  Is this agreeable?

subbu-ai, 2021-01-14:
In the current container configuration, we do not use any -mm value and we do not receive the String values at IoTHub.

{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "-v c:/iiotedge:c:/appdata",
    "--pf=/appdata/pn_4.json",
    "--sf=True",
    "--aa"
  ],
  "HostConfig": {
    "Binds": [
      "c:/iiotedge:c:/appdata"
    ],
    "ExtraHosts": [
      "localhost:127.0.0.1"
    ]
  }
}

What is the best -mm value to send the String values (e.g. OPEN, CLOSE for Valve sensors) ?  Also what is the best format to stream the Status code or Status?

barnstee, 2021-01-14:
Yes, it is acceptable not to specify any heartbeats and there is a default sampling interval in OPC Publisher (1s), if none is specified in the pn.json file. The default can also be overwritten through a command line argument, see oi flag here: https://github.com/Azure/iot-edge-opc-publisher/blob/main/CommandLineArguments.md

barnstee, 2021-01-14:
If you do not supply a messaging mode, PubSub is the default. If the Status is not present, the status is good. You only get status when the status is bad. For the issue where strings don't show up in the telemetry stream, please open a separate GitHub issue and attach your pn.json, command line arguments and resulting telemetry message (use IoT Explorer to read it).

subbu-ai, 2021-01-15:
How to fix this issue:

[08:00:46 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection 
System.Net.HttpListenerException (5): Access is denied.
   at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23

subbu-ai, 2021-01-15:
Frequently I get Communication errors.  How to fix this?

[12:37:38 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://10.0.3.106:49310_cf9df0a198e2a4285a8571bb30a5d3b70493d245
  # Ingestion duration                 :    00:04:28:57 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :         13,232 (0.82/s)
  # Ingress ValueChanges (from OPC)    :      1,137,593 (70.49/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :      1,137,636
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :          2,147
  # Encoder avg Notifications/Message  :            530
  # Encoder avg IoT Message body size  :        172,313 (66%)
  # Encoder avg IoT Chunk (4 KB) usage :           42.1
  # Estimated IoT Chunks (4 KB) per day:        494,281
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :          2,147 (0.13/s)
  # Connection retries                 :              0
 
[12:37:54 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 4: Module dev-meg-iotdevice01_OPCPublisher_1 disconnected due to Communication_Error - now Disconnected_Retrying... 
[12:38:09 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 4: Module dev-meg-iotdevice01_OPCPublisher_1 reconnected due to Connection_Ok. 
[12:38:38 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine] 
  DIAGNOSTICS INFORMATION for          : opc.tcp://10.0.3.106:49310_cf9df0a198e2a4285a8571bb30a5d3b70493d245
  # Ingestion duration                 :    00:04:29:57 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :         13,269 (0.82/s)
  # Ingress ValueChanges (from OPC)    :      1,140,987 (70.44/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :      1,141,030
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :          2,154
  # Encoder avg Notifications/Message  :            530
  # Encoder avg IoT Message body size  :        172,265 (66%)
  # Encoder avg IoT Chunk (4 KB) usage :           42.1
  # Estimated IoT Chunks (4 KB) per day:        494,055
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :          2,154 (0.13/s)
  # Connection retries                 :              0

barnstee, 2021-01-15:
@subbu-ai, like I mentioned above, please create new issues for these, they have nothing to do with your original issue.

barnstee, 2021-01-15:
Closing as the original issue was resolved.

closed 2021-01-15 by barnstee