Azure/iot-edge-opc-publisher-v1 #288: ERR Unable to start metric server

subbu-ai, 2021-01-15:
I am running OPCPublisher version 2.7.199.  I am getting the following error.

How to fix this issue:

[08:00:46 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection
System.Net.HttpListenerException (5): Access is denied.
at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
at System.Net.HttpListener.AddAllPrefixes()
at System.Net.HttpListener.Start()
at Prometheus.MetricServer.StartServer(CancellationToken cancel)
at Prometheus.MetricHandler.Start()
at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23

monikavar, 2021-01-15:
The metrics are exposed using a HttpListener based server. It may throw an access denied exception on Windows if your user does not have the right to open a web server on the specified port. You can use the netsh command to grant yourself the required permissions:
`netsh http add urlacl url=http://+:9702/metrics user=Everyone`

Or, on Windows, please try setting the following in OPC Publisher module's container create options.
`"User": "ContainerAdministrator"`

More details are mentioned in the troubleshooting guide [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/modules/metricscollector.md#troubleshooting).

**Note**: This error should not cause the OPC Publisher module to fail. Would only affect Prometheus metric server.

monikavar, 2021-01-20:
Hope the above guidance has solved your issue.

closed 2021-01-20 by monikavar