Azure/iot-edge-opc-publisher-v1 #289: OPCPublisher disconnected due to Communication_Error

subbu-ai, 2021-01-15:
I am running OPCPublisher 2.7.199 on Windows VM Windows Container.  
Frequently I get Communication errors. How to fix this? Do I loose data?

[12:37:38 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
DIAGNOSTICS INFORMATION for : opc.tcp://10.0.3.106:49310_cf9df0a198e2a4285a8571bb30a5d3b70493d245
Ingestion duration : 00:04:28:57 (dd:hh:mm:ss)
Ingress DataChanges (from OPC) : 13,232 (0.82/s)
Ingress ValueChanges (from OPC) : 1,137,593 (70.49/s)
Ingress BatchBlock buffer size : 0
Encoding Block input/output size : 0 | 0
Encoder Notifications processed : 1,137,636
Encoder Notifications dropped : 0
Encoder IoT Messages processed : 2,147
Encoder avg Notifications/Message : 530
Encoder avg IoT Message body size : 172,313 (66%)
Encoder avg IoT Chunk (4 KB) usage : 42.1
Estimated IoT Chunks (4 KB) per day: 494,281
Outgress Batch Block buffer size : 0
Outgress input buffer count : 0
Outgress input buffer dropped : 0
Outgress IoT message count : 2,147 (0.13/s)
Connection retries : 0

[12:37:54 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 4: Module dev-meg-iotdevice01_OPCPublisher_1 disconnected due to Communication_Error - now Disconnected_Retrying...
[12:38:09 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 4: Module dev-meg-iotdevice01_OPCPublisher_1 reconnected due to Connection_Ok.
[12:38:38 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
DIAGNOSTICS INFORMATION for : opc.tcp://10.0.3.106:49310_cf9df0a198e2a4285a8571bb30a5d3b70493d245
Ingestion duration : 00:04:29:57 (dd:hh:mm:ss)
Ingress DataChanges (from OPC) : 13,269 (0.82/s)
Ingress ValueChanges (from OPC) : 1,140,987 (70.44/s)
Ingress BatchBlock buffer size : 0
Encoding Block input/output size : 0 | 0
Encoder Notifications processed : 1,141,030
Encoder Notifications dropped : 0
Encoder IoT Messages processed : 2,154
Encoder avg Notifications/Message : 530
Encoder avg IoT Message body size : 172,265 (66%)
Encoder avg IoT Chunk (4 KB) usage : 42.1
Estimated IoT Chunks (4 KB) per day: 494,055
Outgress Batch Block buffer size : 0
Outgress input buffer count : 0
Outgress input buffer dropped : 0
Outgress IoT message count : 2,154 (0.13/s)
Connection retries : 0

monikavar, 2021-01-15:
Hey, please provide the full log file with debug log level.

subbu-ai, 2021-01-15:
Please find attached. Debug log level is Default.  Please let me know if you want me to set to different log level and also how to set the log level.

subbu-ai, 2021-01-15:
[OPCPublisher_1_20210114.log](https://github.com/Azure/iot-edge-opc-publisher/files/5820161/OPCPublisher_1_20210114.log)

Log level Default.  Please let me know if you would like to different log level and let me know how to set the new log level.

monikavar, 2021-01-28:
Please provide edgeHub logs as well.

monikavar, 2021-02-05:
EdgeHub re-authenticates all connected clients periodically, and if any of them have expired tokens, those connections are dropped. The IoTHub SDKs create tokens of 1 hour duration by default. So unless you are using your own SDK, or overriding the token duration, that would explain why the client connection is being dropped every hour.

But this behavior is completely normal – what is expected is that the client should immediately re-connect to EdgeHub and resume its activity. If the client is not re-connecting, that would be a bug in the SDK.

From the logs you provided, this is not a bug as reconnection is happening every hour and there is no data loss. Hope this helps.

closed 2021-02-05 by monikavar