Azure/iot-edge-opc-publisher-v1 #292: How to override to send values even they are not changed?

subbu-ai, 2021-01-18:
My OPCPublisher version is 2.7.206.  I understand the default behaviour of the OPCPublisher is send only changed values.  How to send the values even if they are not changed.

subbu-ai, 2021-02-23:
This is important for the tags that has the following behaviour:
1) Setpoint tags that does not change frequently
2) Pi Server implements a concept called compmax parameter (similar to heartbeat).  However, based on my testing, OPCPublisher does not consider the compmax parameter of Pi Server or there is a bug.

subbu-ai, 2021-03-01:
@barnstee  Please change this issue from doc to future request.

barnstee, 2021-03-02:
This is implemented using the heartbeat functionality. Please check the docs here: https://github.com/azure/iot-edge-opc-publisher#configuration-via-configuration-file

ajithkumars, 2021-04-29:
@barnstee This is supposed to work but does not in the latest versions.
In brief it works in version 2.5, 2.6 but no later versions (2.7,2,7.183,2.7.206)

Created an issue here since we are using the full setup: 
https://github.com/Azure/Industrial-IoT/issues/1133

koepalex, 2021-06-10:
Handling this issue as feature request to support Compression of (OSI Soft?) PI Server. Heartbeat bug is tracked in #1133